s60sc/ESP32-CAM_MJPEG2SD #515: SD card cannot read the data

naradashen, 2024-11-07:
[00:00:47.926 startPing] Started ping monitoring - On
[00:00:47.929 startWebServer] Starting web server on port: 80
[00:00:47.930 startWebServer] Remote server certificates not checked
[00:00:47.940 resetWatchDog] WatchDog started using task: ping
[00:00:47.940 getLocalNTP] Using NTP server: pool.ntp.org
[00:00:47.951 WARN getLocalNTP] Not yet synced with NTP
[00:00:47.951 WARN wgetFile] Open failed: /data/common.js
[00:00:47.951 WARN setup] Startup Failure: Check SD card inserted
[00:00:47.962 loop] =============== Total tasks: 12 ===============

[00:00:48.568 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm
[00:00:50.717 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm

Can anyone explain why this happens?.
I uploaded the data folder to the SD card and somehow it cannot read the files in it.
I ran some test codes to see whether the SD is working. It does.
If anyone knows the issue please help me

kepi, 2024-12-04:
I had same problem, SD card not only seems OK everywhere (2GB) but it also worked with AI on the edge.

But I tried another to be sure and everything is working now.

michalrz250, 2024-12-09:
Hey man, I had a similar thing.
Yours seems to be more about the SD card indeed, mine was - the web interface fails to read the SD directories once you have logging to SD enabled. 
After you try to refresh the directory, even the web interface stops loading with a "Nothing with that URI found", and the recordings don't actually make it into the card.

Try disabling Logging to SD.

naradashen, 2024-12-11:
@michalrz250 Is it working now for you?

closed 2024-12-20 by s60sc