s60sc/ESP32-CAM_MJPEG2SD #517: TinyCam Pro stream  Task not created for stream: video, numStreams 1

tomtdl, 2024-11-30:
I have one ESP32-CAM that works on your web interface
AND as a MJPEG stream to TinyCam Pro.

On TinyCam Pro, it uses this request:
http://192.168.1.148:80/sustain?video=1
with the protocol Server push (MJPEG)

Any other ESP32-CAM (I have tried a dozen from different vendors)
will work in your web interface, but fails to stream to TinyCam Pro with the message "Video failed" in TinyCam
AND a message like this in Arduino IDE:
[19:08:49.676 WARN appSpecificSustain consoleHandler] Task not created for stream: video, numStreams 1 

I have scoured the web, looked at your code and the message appears to be coming from line 344 
of streamServer.cpp.  

I am currently selecting:
Board: ESP32 Dev Module
Upload Speed : 921600 (default)
CPU Frequency: 240MHz (default)
Flash Frequency: 40 MHz
Flash Mode: QIO (default)
Flash size: 4MB (default)
Partition Scheme: Huge APP (3MB No OTA/1MB SPIFFS)
Core Debug Level: None
PSRAM: Enabled

The one ESP32-CAM that works can be reprogrammed this way and always works,
though any other will only work on the Web page interface, not the MJPEG stream.

So it appears to be a hardware difference.  All modules have 4MB flash, PSRAM enabled, etc.

I cannot find a fix or this error specific to ESP32-CAM or MJPEG2SD.

I have tried AITHINKER and ESP32 Dev Module with the same results.

What does this tell you ?
WARN appSpecificSustainHandler] Task not created for stream: video, numStreams 1 

Any help is appreciated.  I appreciate all the hard work that went into this project.

tomtdl, 2024-11-30:
This works, however...

Stills: http://192.168.1.148/control?still=1 with Snapshot (JPEG) Protocol

s60sc, 2024-12-01:
did you set `Enable NVR Video stream: /sustain?video=1`
under` Edit Config / Motion`

closed 2024-12-20 by s60sc