s60sc/ESP32-CAM_MJPEG2SD #518: Using ESP32S3 Xiao and can't get audio working

lorde85, 2024-11-30:
I have stubbled to this issue and can-t seem to make it work.

From the documentation it seems straight forward for this device, however, I can't find what's missing.

I've done the following:
-  set #define INCLUDE_AUDIO to true in the appGlobals.h
- confirmed in the UI the values (predefined for XIAO Sense) which are set under Audio button on the configuration web page.
- Raised to 5 in the slider for Microphone Gain.
- Enabled the sustain?audio=1 on the configuration web page.

Can someone shed some light here? What am I missing?

BTW congrats for the work done here, it's been quite helpful to set my DIY babycam.

s60sc, 2024-12-01:
Youve done the right steps
What does the log show?
Have you tested the mic h/w with another app?

random-strings, 2024-12-08:
I had a similar experience the the board esp32s3 xiao sense, which has a built in microphone, the log did not show anything in connection to the microphone. Shame because this board is so much smaller, and you don't have to connect the mic by yourself.

lorde85, 2024-12-09:
> Youve done the right steps What does the log show? Have you tested the mic h/w with another app?

Well not sure what I missed, as I double checked that I enabled audio on the config file.
Thing is that I compiled and uploaded everything again and now it works.
Thanks for your comments and for this great sketch.

closed 2024-12-09 by lorde85