s60sc/ESP32-CAM_MJPEG2SD #519: How to use machine learning library ?

GuillaumeHupi, 2024-12-03:
Hello, 

I have been trying all the features of the project for a few weeks. The project is very complete. Congratulations for all the work done!

I am trying to integrate a machine learning library into the project but I can't do it.

I trained a model on edge impulse, exported it as an arduino library, then copied the unzipped library into the arduino folder.

In the project I modified these two lines:
```
#define INCLUDE_TINYML true // set to true
#define TINY_ML_LIB "Image_classification_inferencing.h" // replace with your lib
```
But I have errors when compiling, I think my model does not have the right input or output format.

I started with an object detection model with FOMO, this one outputs a list of objects and their coordinates. I think this is not what is expected, but rather a classification model. So I also tried to train a classification model. Which allows to know if a person, a car or something else is present on the image. I redid all the steps, but I still have problems related to the output dimensions I think.

Here are the errors I have:

```
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp: In function 'TfLiteStatus tflite::{anonymous}::Prepare(TfLiteContext*, TfLiteNode*)':
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1789:67: error: either all initializer clauses should be designated or none of them should be
 1789 |                                 .channels = input->dims->data[3], 1
      |                                                                   ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1793:68: error: either all initializer clauses should be designated or none of them should be
 1793 |                                 .channels = output->dims->data[3], 1
      |                                                                    ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1795:80: error: either all initializer clauses should be designated or none of them should be
 1795 |     data_dims_t filter_dims = {.width = filter_width, .height = filter_height, 0, 0};
      |                                                                                ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp: In function 'void tflite::{anonymous}::EvalQuantizedPerChannel(TfLiteContext*, TfLiteNode*, const TfLiteConvParams&, const NodeData&, const TfLiteEvalTensor*, const TfLiteEvalTensor*, const TfLiteEvalTensor*, TfLiteEvalTensor*)':
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1883:58: error: either all initializer clauses should be designated or none of them should be
 1883 |                                 .channels = input_depth, 1
      |                                                          ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1887:59: error: either all initializer clauses should be designated or none of them should be
 1887 |                                 .channels = output_depth, 1
      |                                                           ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/conv.cpp:1889:80: error: either all initializer clauses should be designated or none of them should be
 1889 |     data_dims_t filter_dims = {.width = filter_width, .height = filter_height, 0, 0};
      |                                                                                ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp: In function 'void tflite::{anonymous}::EvalQuantizedPerChannel(TfLiteContext*, TfLiteNode*, const TfLiteDepthwiseConvParams&, const NodeData&, const TfLiteEvalTensor*, const TfLiteEvalTensor*, const TfLiteEvalTensor*, TfLiteEvalTensor*)':
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1727:58: error: either all initializer clauses should be designated or none of them should be
 1727 |                                 .channels = input_depth, 1
      |                                                          ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1731:59: error: either all initializer clauses should be designated or none of them should be
 1731 |                                 .channels = output_depth, 1
      |                                                           ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1733:80: error: either all initializer clauses should be designated or none of them should be
 1733 |     data_dims_t filter_dims = {.width = filter_width, .height = filter_height, 0, 0};
      |                                                                                ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp: In function 'TfLiteStatus tflite::{anonymous}::Prepare(TfLiteContext*, TfLiteNode*)':
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1836:67: error: either all initializer clauses should be designated or none of them should be
 1836 |                                 .channels = input->dims->data[3], 1
      |                                                                   ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1840:68: error: either all initializer clauses should be designated or none of them should be
 1840 |                                 .channels = output->dims->data[3], 1
      |                                                                    ^
/Users/guillaumecostes/Documents/Arduino/libraries/Image_classification_inferencing/src/edge-impulse-sdk/tensorflow/lite/micro/kernels/depthwise_conv.cpp:1842:80: error: either all initializer clauses should be designated or none of them should be
 1842 |     data_dims_t filter_dims = {.width = filter_width, .height = filter_height, 0, 0};
      |                                                                                ^

exit status 1
```

Here is the model I trained on edge impulse : 
https://studio.edgeimpulse.com/public/572232/live

I tried with RGB and GREYSCALE pictures, but I got a similar behavior.
Is there a particular input and output format that the model must respect? Is a certain type of model expected?

Thanks in advance.

Guillaume

s60sc, 2024-12-03:
This is due to the edge impulse team not yet updating their software to be compatible with arduino v3.x.
I reported the problem and a fix [here](https://forum.edgeimpulse.com/t/fix-for-compilation-error-when-using-arduino-esp32-core-v3-x/12220)

GuillaumeHupi, 2024-12-04:
Oh OK thank you ! 

After modifiying conv.cpp and depthwise_conv.cpp files like you explained in the issue, it is working. Thank you!

closed 2024-12-04 by GuillaumeHupi