s60sc/ESP32-CAM_MJPEG2SD #520: Wireguard

freakontrol, 2024-12-04:
Hi, I think would be interesting to add Wireguard VPN implementation. 
That would allow to access the camera even in another network! 
I'm already using it with some [esphome](https://github.com/esphome/feature-requests/issues/1444) devices and it works great!

s60sc, 2024-12-04:
I'll take a look in the new year

TheWISPRer, 2024-12-28:
I did this on one of mine, but I think that I lost the code with my old computer. Works great and adds very little overhead. If I can find it, I'll contribute. It did not have any webserver/gui-configurable settings for the connection, but did have a really simple connection config file that the connection info could be placed into. Wireguard is an excellent VPN choice for this application.