s60sc/ESP32-CAM_MJPEG2SD #522: [FEATURE REQUEST]  Need split video files

troublegum, 2024-12-07:
When recording a video, you need to be able to save it in parts, not in a single file.
You need a setting that would allow you to set the ability to save videos to a single file or in parts to different files, limited in size.
This will allow you to lose only part of the video in case of a failure, and not the entire video.

s60sc, 2024-12-07:
Under `Edit Config / Motion` you can set `Max frames in recording`

troublegum, 2024-12-07:
Yes, there is such a setting, but it stops recording when the specified value is reached and I get the message "Auto closed recording after 500 frames"
But the recording should continue, already in a new file.
That is, the video should be recorded in parts, and not in one file.

I have version 9.9.4

s60sc, 2024-12-07:
Recording will continue if motion detected.
If you just want continuous recording, better to stream to NVR

closed 2024-12-20 by s60sc