s60sc/ESP32-CAM_MJPEG2SD #525: SD Card don't work a second time

AlfonzRitzek, 2024-12-23:
After a successful boot, the sd card is no longer recognised on the second boot.

the card can be written and read in the pc, but to be able to use it again in the ESP32 it must be reformatted.

appGlobals changed to
`// User's ESP32 cam board
//#if defined(CONFIG_IDF_TARGET_ESP32)
#define CAMERA_MODEL_AI_THINKER 
//#define CAMERA_MODEL_WROVER_KIT 
//#define CAMERA_MODEL_ESP_EYE 
//#define CAMERA_MODEL_M5STACK_PSRAM 
//#define CAMERA_MODEL_M5STACK_V2_PSRAM 
//#define CAMERA_MODEL_M5STACK_WIDE 
//#define CAMERA_MODEL_M5STACK_ESP32CAM
//#define CAMERA_MODEL_M5STACK_UNITCAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL 
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_TTGO_T_CAMERA_PLUS
//#define AUXILIARY

// User's ESP32S3 cam board
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
//#define CAMERA_MODEL_XIAO_ESP32S3 
//#define CAMERA_MODEL_NEW_ESPS3_RE1_0
//#define CAMERA_MODEL_M5STACK_CAMS3_UNIT
//#define CAMERA_MODEL_ESP32S3_EYE 
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
//#define CAMERA_MODEL_DFRobot_FireBeetle2_ESP32S3
//#define CAMERA_MODEL_DFRobot_Romeo_ESP32S3
//#define CAMERA_MODEL_XENOIONEX
//#define AUXILIARY
//#endif`

s60sc, 2024-12-23:
provide log

AlfonzRitzek, 2024-12-23:
Arduino Serial Monitor:
`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc

[00:00:00.487 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.488 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:00:00.498 logSetup] Compiled with arduino-esp32 v3.0.7
[00:00:00.498 printResetReason] Power on reset
[00:00:00.509 printWakeupReason] Wakeup by reset
[00:00:00.537 WARN prepSD_MMC] SD card mount failed 
[00:00:00.791 setWifiSTA] Wifi Station IP from DHCP
[00:00:00.797 WARN startWifi] SSID  not connected not available, use AP 
[00:00:03.814 startPing] Started ping monitoring - On
[00:00:03.817 startWebServer] Starting web server on port: 80
[00:00:03.817 startWebServer] Remote server certificates not checked
[00:00:03.828 WARN setup] Startup Failure: Check SD card inserted 
[00:00:03.828 loop] =============== Total tasks: 12 ===============

[00:00:08.815 resetWatchDog] WatchDog started using task: ping`

now i can't mount the SD card anymore.
SDcard examle is working fine.

s60sc, 2024-12-23:
from readme:

_If you get error: Startup Failure: Check SD card inserted, or Camera init error 0x105 it is usually a camera board selection issue_

AlfonzRitzek, 2024-12-24:
After consulting the SD Card Chip Datasheet from my ESP32 Module  found out that this modul only supports 4GB sd card.. Changed to a 4gb Card an the Modul is working fine.

closed 2024-12-24 by AlfonzRitzek