s60sc/ESP32-CAM_MJPEG2SD #526: Achieving Stable 100Hz I2C Readings from MPU9250

omridavid, 2024-12-24:
Hi, thank you so much for the amazing code!

I'm running this code on a Xiao ESP32-S3 Sense board and trying to read data from the MPU9250 (accelerometer and gyro) at a stable rate of 100Hz while also recording AVI and streaming via a web GUI.

Configuration:

MPU9250 I2C Pins: I've connected the MPU9250 to I2C pins 5 and 6 and modified the shared I2C pins in the code accordingly.
CSV Structure: I modified the CSV structure to accommodate additional fields for the accelerometer and gyro data.

Changes Made to Achieve Higher Sample Rate:

1. In telemetry.h: Changed `sampleInterval` from 1000 to 1.
2. In telemetry.h: Reduced delay during the collection interval (or commented out this line):
```
// wait for next collection interval
while (millis() - sampleInterval < startTime) delay(1);
```
3. In periphsI2C.cpp: Set a higher sample rate:
`setting.fifo_sample_rate = FIFO_SAMPLE_RATE::SMPL_1000HZ;`
4. In periphsI2C.cpp: Commented out this line:
`while (!mpu9250.update() && millis() - mpuWait < SENSOR_TIMEOUT) delay(10);`
5. In periphsI2C.cpp: Increased the I2C clock speed with:
`Wire.setClock(400000);`
6. In appGlobals.h: Increased the priority of telemetry task with:
`#define TELEM_PRI 7`

Issue: Despite these modifications, the reading rates are inconsistent, sometimes dropping below 100Hz and at other times exceeding it.

Request for Suggestions: Do you have any suggestions for achieving stable 100Hz readings from the MPU9250? Any tips on further optimizing the code or managing task priorities would be greatly appreciated.

Thank you in advance for your help!

Best,
Om

s60sc, 2024-12-24:
Thanks
This app is resource intensive so its asking a lot of the esp to sample at exactly 100Hz whilst the rest of the app is running.
You could look at running the telemetry task on an [Auxiliary Board](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#auxiliary-board)

omridavid, 2024-12-24:
Thank you! 
I understand. 
Unfortunately, incorporating an auxiliary board isn’t possible in my case.
Maybe do you have any suggestion how can I increase the reading rate to ensure it consistently stays above 100Hz?

s60sc, 2024-12-24:
That rate is not feasible with this app

omridavid, 2025-01-06:
I was able to achieve a telemetry rate close to 100Hz by following these steps:

1. Increased the telemetry task priority to 6.
2. Set the I2C clock speed to 400kHz.
3. Configured the IMU sample rate to its maximum.
4. Included wait times that take into account the sample time.
5. Pinned the telemetry task to core 1 (though I'm unsure if this was necessary).

Thank you very much for your code and your assistance!

closed 2025-01-22 by s60sc