s60sc/ESP32-CAM_MJPEG2SD #528: XAOI ESP32S3 Sense: Enable onboard mic?

mbell85, 2024-12-31:
Apologies for my ignorance here. I really enjoy this project and thank you so much! How would I enable the onboard Mic on the esp32S3 Sense for recording audio? The microphone is greyed out in the Web UI, so I assume I need to make some change inside the IDE.

s60sc, 2025-01-01:
Thanks
The greyed out microphone on the Web UI is for the browser mic, not esp mic
Use of esp mic covered [here](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#audio-recording)

PD1963, 2025-01-05:
I also experience this issue with the XAOI ESP32S3 Sense.
The Speaker icon button on the web page that can be used to listen to the microphone from the browser does not show up. I hear no sound with the mic gain on 10.

I have set '#define INCLUDE_AUDIO' to true.
Also in video recordings the mic works because sound is recorded (with gain 10).

Thank you in advance for your help!

closed 2025-01-10 by s60sc