s60sc/ESP32-CAM_MJPEG2SD #530: [NEW FEATURE SOON] RTSP Video, Audio, Subtitles Server

rjsachse, 2025-01-04:
I was creating an RTSP server for this project [ESP32-CAM_MJPEG2SD](https://github.com/s60sc/ESP32-CAM_MJPEG2SD) (Not yet implemented) and realised it would be even more useful as a library so it can be easily integrated into other projects. The RTSP Server library enables you to stream Video, Audio, and Subtitles all in one RTSP stream. It supports UDP, TCP, and Multicast, though currently is limited to one client at a time. TCP is slower than UDP, Unicast, or Multicast, but all options are available for flexibility. This initial release was quickly put together before I returned to work, with plans for further optimization. Only tested on my own esp32s3 custom board.

Try out the example and explore the capabilities of the RTSP Server library here: [RTSPServer](https://github.com/rjsachse/RTSPServer)

rjsachse, 2025-01-08:
pull request submitted

s60sc, 2025-01-12:
v10.5.1 modifies rtsp.cpp to use app's sustain tasks rather than library task

rjsachse, 2025-01-19:
I tried to add to sustain task but failed only issue I'm getting now is if you change the resolution VLC freaks out now, but looking at the packets using wireshark new resolution is correct, so a strange 1

rjsachse, 2025-01-26:
i noticed you created a task for subtitles. if using a task just need to set delay for 1 sec interval and no need for time as the timer just calls a function every 1 sec

s60sc, 2025-01-26:
using delay on its own causes drift unless the time taken by the task and esp between delays is calculated to reduce each delay for precise timing.

rjsachse, 2025-02-01:
you make a good point, thanks for the tip

closed 2025-02-01 by rjsachse