s60sc/ESP32-CAM_MJPEG2SD #531: No video stream - log shows Failed to get camera frame warning

averter, 2025-01-04:
Hi all,

First post in this great project. After flashing the firmware my ESP32-CAM works as expected: connects to the Wifi, the interface displays correctly, etc. However, I cannot see any video stream, do recordings or stills, and I get a `Failed to get camera frame` warning (log below).

````
00:00:00.519 logSetup] Setup RAM based log, size 7168, starting from 0
[00:00:00.520 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:00:00.530 logSetup] Compiled with arduino-esp32 v3.0.7
[00:00:00.530 printResetReason] Power on reset
[00:00:00.541 printWakeupReason] Wakeup by reset
[00:00:00.588 infoSD] SD card type SDHC, Size: 7.5GB
[00:00:00.588 listFolder] Sketch size 1.6MB
[00:00:00.915 listFolder] File: /data/configs.txt, size: 7333 bytes
[00:00:00.916 listFolder] SD_MMC: 128KB used of 7.5GB
[00:00:00.916 loadConfig] Load config
[00:00:01.197 updateAppStatus] Enabling motion detection
[00:00:01.204 setup] PSRAM size: 4.0MB
[00:00:01.397 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:01.646 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.670 WARN startWifi] SSID not connected not available, use AP
[00:00:01.673 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_F02C9CBBBC2C started, use 'http://192.168.4.1' to connect
[00:00:04.889 startPing] Started ping monitoring - On
[00:00:04.893 startWebServer] Starting web server on port: 80
[00:00:04.894 startWebServer] Remote server certificates not checked
[00:00:04.904 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:08.904 WARN prepRecording] Failed to get camera frame
[00:00:08.904 checkMemory] Free: heap 117012, block: 86004, min: 116628, pSRAM 1483244
[00:00:08.915 loop] =============== Total tasks: 14 ===============

[00:00:09.890 resetWatchDog] WatchDog started using task: ping
[00:01:55.425 onWiFiEvent] WiFi AP client connection
[00:02:31.280 onWiFiEvent] WiFi AP client disconnection
[00:02:31.282 onWiFiEvent] WiFi AP client connection
[00:02:48.502 savePrefs] Saved preferences
[00:02:48.529 saveConfigVect] Config file saved
[00:02:48.620 doRestart] Controlled restart: User requested restart
[00:02:50.623 onWiFiEvent] Wifi Station stopped mynetworkname
[00:02:50.623 onWiFiEvent] WiFi AP client disconnection
[00:02:51.701 logSetup] Setup RAM based log, size 7168, starting from 1961

[00:02:51.701 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:02:51.711 logSetup] Compiled with arduino-esp32 v3.0.7
[00:02:51.712 printResetReason] Software reset via esp_restart
[00:02:51.722 printWakeupReason] Wakeup by reset
[00:02:51.769 infoSD] SD card type SDHC, Size: 7.5GB
[00:02:51.770 listFolder] Sketch size 1.6MB
[00:02:52.096 listFolder] File: /data/configs.txt, size: 7351 bytes
[00:02:52.097 listFolder] SD_MMC: 128KB used of 7.5GB
[00:02:52.097 loadConfig] Load config
[00:02:52.377 updateAppStatus] Enabling motion detection
[00:02:52.384 setup] PSRAM size: 4.0MB
[00:02:52.577 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:02:52.853 setWifiSTA] Wifi Station IP from DHCP
.[00:02:52.889 onWiFiEvent] WiFi Station connection to mynetworkname, using hostname: ESP-CAM_MJPEG_F02C9CBBBC2C
[00:02:53.129 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.191' to connect
[00:02:53.436 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_F02C9CBBBC2C started, use 'http://192.168.4.1' to connect
[00:02:56.647 startWifi] Wifi stats for mynetworkname - signal strength: -54 dBm; Encryption: WPA2_PSK; channel: 1
[00:02:56.649 startPing] Started ping monitoring - On
[00:02:56.663 startWebServer] Starting web server on port: 80
[00:02:56.663 startWebServer] Remote server certificates not checked
[00:02:56.663 startSustainTasks] Started 1 HTTP sustain tasks
[00:02:56.666 resetWatchDog] WatchDog started using task: ping
[00:02:56.674 getLocalNTP] Using NTP server: pool.ntp.org
[00:02:56.685 WARN getLocalNTP] Not yet synced with NTP
[00:03:00.334 onWiFiEvent] WiFi AP client connection
[00:03:00.674 WARN prepRecording] Failed to get camera frame
[00:03:00.674 checkMemory] Free: heap 59880, block: 30708, min: 57220, pSRAM 1483196
[00:03:00.685 loop] =============== Total tasks: 14 ===============

[00:03:02.136 savePrefs] Saved preferences
[00:03:02.165 saveConfigVect] Config file saved
[00:03:02.274 doRestart] Controlled restart: User requested restart
[00:03:04.280 onWiFiEvent] WiFi Station disconnected
[00:03:04.280 onWiFiEvent] Wifi Station stopped mynetworkname
[00:03:04.281 onWiFiEvent] WiFi AP client disconnection
[00:03:05.447 logSetup] Setup RAM based log, size 7168, starting from 4247

[00:03:05.448 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:03:05.458 logSetup] Compiled with arduino-esp32 v3.0.7
[00:03:05.458 printResetReason] Software reset via esp_restart
[00:03:05.469 printWakeupReason] Wakeup by reset
[00:03:05.566 infoSD] SD card type SDHC, Size: 7.5GB
[00:03:05.567 listFolder] Sketch size 1.6MB
[00:03:05.894 listFolder] File: /data/configs.txt, size: 7351 bytes
[00:03:05.897 listFolder] File: /data/common.js, size: 0 bytes
[00:03:05.897 listFolder] SD_MMC: 128KB used of 7.5GB
[00:03:05.908 loadConfig] Load config
[00:03:06.177 updateAppStatus] Enabling motion detection
[00:03:06.184 setup] PSRAM size: 4.0MB
[00:03:06.377 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:03:06.627 setWifiSTA] Wifi Station IP from DHCP
.[00:03:06.769 onWiFiEvent] WiFi Station connection to mynetworkname, using hostname: ESP-CAM_MJPEG_F02C9CBBBC2C
.[00:03:07.251 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.191' to connect
[00:03:07.657 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_F02C9CBBBC2C started, use 'http://192.168.4.1' to connect
[00:03:10.869 startWifi] Wifi stats for mynetworkname - signal strength: -56 dBm; Encryption: WPA2_PSK; channel: 1
[00:03:10.870 startPing] Started ping monitoring - On
[00:03:10.878 resetWatchDog] WatchDog started using task: ping
[00:03:10.881 getLocalNTP] Using NTP server: pool.ntp.org
[00:03:10.892 WARN getLocalNTP] Not yet synced with NTP
[00:03:10.892 startWebServer] Starting web server on port: 80
[00:03:10.903 startWebServer] Remote server certificates not checked
[00:03:10.903 startSustainTasks] Started 1 HTTP sustain tasks
[00:03:11.273 wgetFile] Downloading /data/common.js from /s60sc/ESP32-CAM_MJPEG2SD/master/data/common.js
[00:03:12.045 wgetFile] Downloaded /data/common.js, size 46192 bytes
[00:03:12.436 wgetFile] Downloading /data/MJPEG2SD.htm from /s60sc/ESP32-CAM_MJPEG2SD/master/data/MJPEG2SD.htm
[12:50:52.280 wgetFile] Downloaded /data/MJPEG2SD.htm, size 97KB
[12:50:53.721 WARN prepRecording] Failed to get camera frame
[12:50:53.722 checkMemory] Free: heap 114820, block: 40948, min: 53884, pSRAM 1483640
[12:50:53.732 loop] =============== Total tasks: 14 ===============

[12:51:14.956 wsHandler] Websocket connection: 53
[12:51:19.692 getLocalNTP] Using NTP server: pool.ntp.org
[12:51:19.693 showLocalTime] Got current time from NTP: 04/01/2025 12:51:19 with tz: GMT0
````
I sometimes also get the following warning although it is probably unrelated
````
[12:59:27.104 WARN wsHandler] closing connection, as newer Websocket on 54 
````
The only issue that I've found where this warning shows up [is this](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/172) but it is related to an SD card malfunction which is not the case here.

 If I search it online the warning that everyone gets is different, `Failed to get camera frame warning on time`, and therefore probably unrelated.

Any thoughts? Other things worth checking? Thanks.

s60sc, 2025-01-04:
Problem is reported by espressif camera library, normally due to camera hardware fault.
Check camera ribbon cable seated properly and undamaged, try a different camera or board etc

averter, 2025-01-04:
I admit to have struggled a bit with the camera ribbon to put in place on the first time. 
After checking it seems to be undamaged and connected well, so I'll try with another ESP32-CAM and report back.

closed 2025-01-10 by s60sc