s60sc/ESP32-CAM_MJPEG2SD #532: not compiling on new update.

josef2600, 2025-01-05:
sorry to bug you, but since your new update from 9 hours ago , now  it doesn't compile at all. i get this:
```
C:/Users/Josef/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `devSetup()'
C:/Users/Josef/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o: in function `setup()':
D:\micro\2024\esp32-audio cam\Camera\2\ESP32-CAM_MJPEG2SD/ESP32-CAM_MJPEG2SD.ino:20:(.text._Z5setupv+0x40): undefined reference to `devSetup()'
collect2.exe: error: ld returned 1 exit status

```

s60sc, 2025-01-05:
doh, forgot to uncomment it - updated
Only relevant to my dev environment

josef2600, 2025-01-05:
sorry to do it, but i have another weird problem. i have a custom board. i have made all relative pins in the code and made up a custom name for it.
at first, i thought my connections are bad. but today i checked more carefully and i dont think it is the connections. this is the debug:
```
00:00:01.008 logSetup] =============== ESP-CAM_MJPEG 10.4.3 ===============
[00:00:01.021 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[00:00:01.021 boardInfo] Flash 16.0MB, mode QIO @ 80Mhz
[00:00:01.022 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[00:00:01.032 logSetup] Compiled with arduino-esp32 v3.1.0
[00:00:01.032 printResetReason] Power on reset
[00:00:01.042 printWakeupReason] Wakeup by reset
[00:00:01.142 setup] Selected board Dirty_CAMERA_ESP32S3_CAM_v1
[00:00:01.349 infoSD] SD card type SDHC, Size: 14.5GB
[00:00:01.349 listFolder] Sketch size 1.5MB
[00:00:01.492 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[00:00:01.496 listFolder] File: /data/common.js, size: 46192 bytes
[00:00:01.500 listFolder] File: /data/MJPEG2SD.htm, size: 97KB
[00:00:01.501 listFolder] SD_MMC: 6.8MB used of 14.4GB
[00:00:01.511 loadConfig] Load config
[00:00:01.732 checkConfigFile] Created /data/configs.txt from local store
[00:00:01.845 updateAppStatus] Enabling motion detection
[00:00:02.037 prepCam] Camera init OK for OV2640
[00:00:02.241 setWifiSTA] Wifi Station IP from DHCP
[00:00:02.256 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_34758ABD4D74
[00:00:02.257 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_34758ABD4D74 started, use 'http://192.168.4.1' to connect
[00:00:02.257 WARN startWifi] SSID  not connected not available, use AP 
[00:00:02.279 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_3.local
[00:00:06.092 startPing] Started ping monitoring - On
[00:00:06.094 startWebServer] Starting web server on port: 80
[00:00:06.095 startWebServer] Remote server certificates not checked
[00:00:06.105 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:11.092 resetWatchDog] WatchDog started using task: ping
[00:00:14.105 ERROR @ mjpeg2sd.cpp:725] Failed to get camera frame - check camera hardware 
[00:00:14.105 checkMemory]  Free: heap 167736, block: 106484, min: 167736, pSRAM 5655512
[00:00:14.116 loop] =============== Total tasks: 16 ===============

[00:00:57.688 onWiFiEvent] WiFi AP client connection
[00:01:02.699 wsHandler] Websocket connection: 53
[11:58:20.224 showLocalTime] Got current time from browser: 05/01/2025 11:58:20 with tz: GMT0
[11:58:36.581 wsHandler] Websocket connection: 53
[11:58:45.396 wsHandler] Websocket connection: 53
[11:59:22.858 WARN wsHandler] closing connection, as newer Websocket on 52 
[11:59:22.859 wsHandler] Websocket connection: 52
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

```
i wanted to check if my pins are connected, like check the i2c address. but it does it `prepCam] Camera init OK for OV2640`.
but then it makes error on connection to camera. and if i try to stream, it crashes. do you have any suggestion for me?
and there is another weird thing! not related to your code, but i have seen it few times. i have connected a 120 Mhz flash to it (well, the flash can handle 133Mhz) and the settings are ok. but i get 80Mhz for it. and checked on few different boards. do you have any experience on this? thank you very much. the PSRAM is internal.

s60sc, 2025-01-05:
Can you send the decoded stack trace using [Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder)

josef2600, 2025-01-05:
thank you. i don't know what it is, but i will try. but out of curiosity i disconnected the camera to see what happens. and it made error for initializing it.
`:00:06.254 WARN setup] Startup Failure: Camera init error 0x105 on Dirty_CAMERA_ESP32S3_CAM_v1 `

s60sc, 2025-01-05:
That is expected error. 
Exception Decoder is must have when developing

josef2600, 2025-01-05:
so this is that!
the first crash:

```
Decoding stack results
0x403768c2: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 463
0x403802cd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x403802cd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x40386f41: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x40386f41: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x40383797: xTaskGenericNotify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c line 5906
0x40383797: xTaskGenericNotify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c line 5906
0x42009729: appSpecificSustainHandler(httpd_req*) at D:\micro\2024\SDR Home Automation\esp32-audio cam\Camera\ESP32-CAM_MJPEG2SD/streamServer.cpp line 341
0x42009729: appSpecificSustainHandler(httpd_req*) at D:\micro\2024\SDR Home Automation\esp32-audio cam\Camera\ESP32-CAM_MJPEG2SD/streamServer.cpp line 341
0x420649e9: httpd_uri at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_uri.c line 329
0x420649e9: httpd_uri at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_uri.c line 329
0x42063a20: httpd_req_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_parse.c line 663
0x42063a20: httpd_req_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_parse.c line 663
0x420640e2: httpd_sess_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 427
0x42062d64: httpd_process_session at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 267
0x42062d64: httpd_process_session at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 267
0x4210bece: httpd_sess_enum at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 50
0x4210bece: httpd_sess_enum at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 50
0x42062f0f: httpd_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 317
0x40380f16: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
0x40380f16: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139

 line 329
0x420649e9: httpd_uri at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_uri.c line 329
0x42063a20: httpd_req_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_parse.c line 663
0x42063a20: httpd_req_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_parse.c line 663
0x420640e2: httpd_sess_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 427
0x42062d64: httpd_process_session at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 267
0x42062d64: httpd_process_session at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 267
0x4210bece: httpd_sess_enum at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 50
0x4210bece: httpd_sess_enum at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 50
0x42062f0f: httpd_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 317
0x40380f16: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
0x40380f16: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139

```
the second crash:

```
Decoding stack results
0x403768c2: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 463
[... cut ...]
0x4210bece: httpd_sess_enum at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_sess.c line 50
0x42062f0f: httpd_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_main.c line 317
0x40380f16: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139

```

josef2600, 2025-01-05:
> That is expected error. Exception Decoder is must have when developing

i know! i was making sure it happens and it is not saying it is connected.

josef2600, 2025-01-05:
to be more specific, i created a `my_headd.h` file and added `#include "my_headd.h"` to all files.
it is that:
```
#ifndef myParameters_h
#define myParameters_h

#define Dirty_CAMERA_ESP32S3_CAM
#define CAM_CSI_HSYNC  1
#define CAM_CSI_PCLK  2
#define CAM_CSI_D7  7
#define CAM_CSI_D6  6
#define CAM_CSI_D5  5
#define CAM_CSI_D4  4
#define CAM_CSI_D3  47
#define CAM_CSI_D2  40
#define CAM_CSI_D1  39
#define CAM_CSI_D0  38
#define CAM_CSI_MCLK  0
#define CAM_TWI_SCK  8
#define CAM_TWI_SDA  18
#define CAM_CSI_VSYNC  17
#define CAM_CAM_PWR  15

#define DATA0_SD1P  41
#define DATA1_SD1P  45
#define DATA2_SD1P  20
#define DATA3_SD1P  9
#define CMD_SD1P  21
#define CLK_SD1P  19

#define my_MIC_WS  11
#define my_MIC_SCK  10
#define my_MIC_DA_SD1P  46

#define N_LED_P_FLASH  16

#define io14P  14
#define io13P  13
#define io12P  12
#define io3P  3

#define CAM_BOARD  "Dirty_CAMERA_ESP32S3_CAM_v1"
#define PWDN_GPIO_NUM  CAM_CAM_PWR
#define RESET_GPIO_NUM  -1
#define XCLK_GPIO_NUM  CAM_CSI_MCLK
#define SIOD_GPIO_NUM  CAM_TWI_SDA
#define SIOC_GPIO_NUM  CAM_TWI_SCK

#define Y9_GPIO_NUM       CAM_CSI_D7
#define Y8_GPIO_NUM       CAM_CSI_D6
#define Y7_GPIO_NUM       CAM_CSI_D5
#define Y6_GPIO_NUM       CAM_CSI_D4
#define Y5_GPIO_NUM       CAM_CSI_D3
#define Y4_GPIO_NUM       CAM_CSI_D2
#define Y3_GPIO_NUM       CAM_CSI_D1
#define Y2_GPIO_NUM       CAM_CSI_D0
#define VSYNC_GPIO_NUM    CAM_CSI_VSYNC
#define HREF_GPIO_NUM     CAM_CSI_HSYNC
#define PCLK_GPIO_NUM     CAM_CSI_PCLK

#define LED_GPIO_NUM     N_LED_P_FLASH
//#define TRIGGER  15 // TRIGER FROM PIR OR RADAR
// Define SD Pins
#define SD_MMC_CLK CLK_SD1P 
#define SD_MMC_CMD CMD_SD1P
#define SD_MMC_D0 DATA0_SD1P
#define SD_MMC_D1 DATA1_SD1P
#define SD_MMC_D2 DATA2_SD1P
#define SD_MMC_D3 DATA3_SD1P
// Define I2S Mic Pins
#define I2S_SD   my_MIC_DA_SD1P  // I2S Microphone
#define I2S_WS   my_MIC_WS
#define I2S_SCK  my_MIC_SCK

/*
Do not use pins: PSRAM
37-36-35-48-47-33-34-

*/
#endif

```
and in `appGlobals.h` i commented out all boards and added this `#define Dirty_CAMERA_ESP32S3_CAM` on top.
and in `camera_pins.h` i added this in the middle :
```
#elif defined(Dirty_CAMERA_ESP32S3_CAM)
#define CAM_BOARD  "Dirty_CAMERA_ESP32S3_CAM_v1"

```
and nothing else.

s60sc, 2025-01-05:
In your log, the camera is prepped ok (via I2C):
`[00:00:02.037 prepCam] Camera init OK for OV2640`
but cant get data from camera:
`[00:00:14.105 ERROR @ mjpeg2sd.cpp:725] Failed to get camera frame - check camera hardware `
so pin assignment problem or ribbon cable seating etc

This causes the subsequent exception,  so I will change the code so that above error will be a startup failure.

s60sc, 2025-01-05:
`mjpeg2sd.cpp` updated to report startup failure on camera fault

josef2600, 2025-01-05:
thank you. i just burnt my chip! i don't know where did i put the other one. so i am closing this. 
and thank you for your fast responses.

josef2600, 2025-01-08:
hi! after extensive research,
about 120MHZ flash - psram.
i found out the 120MHZ is bullshit! it is unusable and unstable.
1. if the temperature goes up or down more than 20 degrees Celsius, it crashes. the data will get corrupted. on both flash and psram.
2. by default, if we use 80mhz, it works on DDR and 8-bit. meaning, double data rate, meaning 1 data transfer on upper clock and one on down. meaning double the speed, meaning the real speed is 160 MHZ and octal (8-bit). but if we use 120Mhz, we lose the double and it will be on quad for psram and double for flash. meaning we lose double rate data transfer speed and lose 8 or 4-bit data transfer on booth psram and flash.
3. in general, it is experimental. and by some tests, it is getting less speed for read and write on flash too. and remember, the data bit rate will be on half on both psram and flash. meaning if we have a quad flash, it will only work on double bit rate. and the same goes for psram too. and besides, all flash chips they use, has a maximum speed of 103 MHZ clock. and they even can’t be clocked with 120 MHz. that is why i made so many changes.

so, i will not be using that and i hope it was fun for you too. that is why they are so faster than old generation.
best regards, Josef.