s60sc/ESP32-CAM_MJPEG2SD #533: error: 'configRUN_TIME_COUNTER_TYPE' was not declared in this scope

jmccarty1974, 2025-01-06:
When trying to compile I recieve the above error. I have tried downloading the code serveral times but keep getting the same error on multiple computers. runCounter is not declared? How an what to I need to declare it as?

s60sc, 2025-01-06:
update arduino ide to latest esp core v3.1.0

jmccarty1974, 2025-01-06:
> update arduino ide to latest esp core v3.1.0

Yeah I fixed that part. I had to download from the releases page and now it compiles. Flashed my board but when I enter 192.168.4.1 to access it I get the error Startup Failure: Camera init error 0x105 on CAMERA_MODEL _AI_THINKER. Any idea's?

jmccarty1974, 2025-01-06:
I believe after much research I have a bad board. Thanks for your help.

closed 2025-01-10 by s60sc