s60sc/ESP32-CAM_MJPEG2SD #537: XAOI ESP32S3 Sense: not able to listen audio via browser

PD1963, 2025-01-10:
I'm experiencing an issue with the XAOI ESP32S3 Sense and listening to audio via the browser.
The Speaker icon button on the web page that can be used to listen to the microphone does not appear. I hear no sound with the mic gain set to gain 10 (in the settings). I have set '#define INCLUDE_AUDIO' to true.

However in video recordings audio is recorded, so the mic works (gain 10).

Thank you in advance for your help!

josef2600, 2025-01-10:
hi. why didn't you say how you handled it and solved it? and why didn't you closed it? it will be nice for others to know it too! 
but for others to know, it was written on the readme.

closed 2025-01-10 by s60sc

PD1963, 2025-01-11:
I'm sorry I made a mistake, I saw no speaker button. Thought it would be fixed with a fresh install. Thats the case but when I use the Speaker icon button on the web page I hear no sound.

I have set #define INCLUDE_AUDIO to true.
Microfone gane is 10.
Recording audio does work.