s60sc/ESP32-CAM_MJPEG2SD #541: WiFi and settings are too unstable.

josef2600, 2025-01-13:
WiFi and settings are too unstable. it constantly hangs, sometimes it restarts, specially at power on, it connects to WiFi but cannot connect then when connecting, it suddenly drupes the connection and try to connect again. it is very messy.
```
[00:00:00.051 logSetup] =============== ESP-CAM_MJPEG 10.5.1 ===============
[00:00:00.063 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[00:00:00.064 boardInfo] Flash 16.0MB, mode QIO @ 80Mhz
[00:00:00.064 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[00:00:00.075 logSetup] Compiled with arduino-esp32 v3.1.1
[00:00:00.075 printResetReason] Power on reset
[00:00:00.085 printWakeupReason] Wakeup by reset
[00:00:00.085 setup] Selected board CAMERA_MODEL_ESP32S3_EYE_FREENOVE
[00:00:00.203 infoSD] SD card type SDHC, Size: 29.8GB @ 40MHz, using 4 bit mode
[00:00:00.203 listFolder] Sketch size 1.5MB
[00:00:00.345 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[00:00:00.348 listFolder] File: /data/common.js, size: 46192 bytes
[00:00:00.351 listFolder] File: /data/MJPEG2SD.htm, size: 97KB
[00:00:00.352 listFolder] SD_MMC: 6.8MB used of 29.8GB
[00:00:00.362 loadConfig] Load config
[00:00:00.390 checkConfigFile] Created /data/configs.txt from local store
[00:00:00.504 updateAppStatus] Enabling motion detection
[00:00:01.250 prepCam] Camera init OK for OV5640
[00:00:01.603 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.621 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_984B014AEC24 started, use 'http://192.168.4.1' to connect
[00:00:01.622 WARN startWifi] SSID  not connected not available, use AP 
[00:00:01.634 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[00:00:05.448 startPing] Started ping monitoring - On
[00:00:05.450 startWebServer] Starting web server on port: 80
[00:00:05.451 startWebServer] Remote server certificates not checked
[00:00:05.461 startSustainTasks] Started 1 sustain tasks
[00:00:05.482 updateAppStatus] Enabling motion detection
[00:00:05.546 prepRecording] To record new AVI, do one of:
[00:00:05.547 prepRecording] - press Start Recording on web page
[00:00:05.547 prepRecording] - move in front of camera
 
[00:00:05.557 prepRecording] Camera model OV5640 ready @ 20MHz
[00:00:05.558 checkMemory] Setup Free: heap 160372, block: 98292, min: 160328, pSRAM 3257604
[00:00:05.568 loop] =============== Total tasks: 18 ===============

[00:00:10.448 resetWatchDog] WatchDog started using task: ping
[00:00:22.505 onWiFiEvent] WiFi AP client connection
[00:00:27.944 wsHandler] Websocket connection: 53
[19:46:11.224 showLocalTime] Got current time from browser: 13/01/2025 19:46:11 with tz: GMT0
[19:46:23.915 WARN updateAppStatus] Motion detection disabled as frame size P_FHD is too large 
[19:46:25.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:27.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:29.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:31.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:33.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:35.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:37.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:37.432 updateAppStatus] Disabling motion detection
[19:46:39.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:40.156 savePrefs] Saved preferences
[19:46:40.176 saveConfigVect] Config file saved 
[19:46:41.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:43.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:45.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:47.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:46:49.327 WARN checkMotion] Frame size P_FHD too large for motion detection 
[19:47:05.056 onWiFiEvent] WiFi AP client disconnection
[19:48:30.453 isNight] Night time
[19:49:22.707 isNight] Day time
[19:49:32.215 onWiFiEvent] WiFi AP client connection
[19:50:11.344 savePrefs] Saved preferences
[19:50:13.307 onWiFiEvent] WiFi AP client disconnection
[19:50:15.345 WARN saveConfigVect] Failed to save to configs file 
[19:50:23.661 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[19:50:23.661 setWifiSTA] Wifi Station IP from DHCP
..........[19:50:28.677 WARN startWifi] SSID josef_wifi not connected unable to connect 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
[... 89 lines cut ...]
[00:01:01.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:03.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:05.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:07.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:09.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:11.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:13.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:15.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:15.879 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[00:01:15.879 setWifiSTA] Wifi Station IP from DHCP
[00:01:15.958 onWiFiEvent] WiFi Station disconnected
[00:01:15.958 WARN sendChunks] Failed to send /data/MJPEG2SD.htm to browser 
..[00:01:16.960 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
.[00:01:17.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
....[00:01:19.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
....[00:01:21.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:21.485 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[00:01:21.487 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[00:01:23.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:25.000 WARN checkMotion] Frame size P_FHD too large for motion detection 
[00:01:25.300 startWifi] Wifi stats for josef_wifi - signal strength: -38 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[00:01:25.479 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect
```

per my experiant, it usaully is from not putting a delay after the wifi setup. you have to put at least 600ms to 1000ms after the wifi setup (wifi.begin).

josef2600, 2025-01-13:
so, i tried this:
in file `utils.cpp` in `bool startWifi(bool firstcall)` , i changed delay from 100 to 1000 , and in `static void setWifiSTA()` after `WiFi.config(_ip, _gw, _sn, _ns1); // need DNS for SNTP` , i put `delay(1000);`  , and after `WiFi.begin(ST_SSID, ST_Pass);` i put `delay(1000);` too. 
but it didnt fix it.
```
[20:08:33.283 savePrefs] Saved preferences
[20:08:33.303 saveConfigVect] Config file saved 
[20:08:39.033 doRestart] Controlled restart: User requested restart 
[20:08:39.534 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[20:08:39.535 setWifiSTA] Wifi Station IP from DHCP
.[20:08:41.035 onWiFiEvent] Wifi Station stopped josef_wifi
[20:08:41.035 onWiFiEvent] WiFi AP client disconnection
......ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037ca4e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac

[20:08:44.138 logSetup] Setup RAM based log, size 7168, starting from 3868

[20:08:44.138 logSetup] =============== ESP-CAM_MJPEG 10.5.1 ===============
[20:08:44.151 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[20:08:44.151 boardInfo] Flash 16.0MB, mode QIO @ 80Mhz
[20:08:44.151 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[20:08:44.162 logSetup] Compiled with arduino-esp32 v3.1.1
[20:08:44.162 printResetReason] Software reset via esp_restart
[20:08:44.172 printWakeupReason] Wakeup by reset
[20:08:44.172 setup] Selected board CAMERA_MODEL_ESP32S3_EYE_FREENOVE
[20:08:44.228 infoSD] SD card type SDHC, Size: 29.8GB @ 40MHz, using 4 bit mode
[20:08:44.229 listFolder] Sketch size 1.5MB
[20:08:44.370 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[20:08:44.374 listFolder] File: /data/common.js, size: 46192 bytes
[20:08:44.377 listFolder] File: /data/MJPEG2SD.htm, size: 97KB
[20:08:44.380 listFolder] File: /data/configs.txt, size: 7702 bytes
[20:08:44.390 listFolder] SD_MMC: 6.8MB used of 29.8GB
[20:08:44.391 loadConfig] Load config
[20:08:44.623 updateAppStatus] Disabling motion detection
[20:08:45.370 prepCam] Camera init OK for OV5640
[20:08:46.627 setWifiSTA] Wifi Station IP from DHCP
..[20:08:48.432 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
........[20:08:52.714 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_984B014AEC24 started, use 'http://192.168.4.1' to connect
[20:08:52.714 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[20:08:52.727 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[20:08:56.340 startWifi] Wifi stats for josef_wifi - signal strength: -46 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[20:08:56.341 startPing] Started ping monitoring - On
[20:08:56.354 startWebServer] Starting web server on port: 80
[20:08:56.354 startWebServer] Remote server certificates not checked
[20:08:56.365 startSustainTasks] Started 1 sustain tasks
[20:08:56.385 updateAppStatus] Disabling motion detection
[20:08:56.450 prepRecording] To record new AVI, do one of:
[20:08:56.450 prepRecording] - press Start Recording on web page
 
[20:08:56.450 prepRecording] Camera model OV5640 ready @ 20MHz
[20:08:56.461 checkMemory] Setup Free: heap 160100, block: 98292, min: 160052, pSRAM 3254344
[20:08:56.471 loop] =============== Total tasks: 18 ===============

[20:08:56.648 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect
[20:09:01.342 resetWatchDog] WatchDog started using task: ping
[20:09:01.342 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[20:09:01.353 setWifiSTA] Wifi Station IP from DHCP
[20:09:01.385 onWiFiEvent] WiFi Station disconnected
[20:09:01.909 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
..........[20:09:07.413 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[20:09:07.414 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[20:09:11.238 startWifi] Wifi stats for josef_wifi - signal strength: -46 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[20:09:11.286 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect
[20:09:23.032 isNight] Night time
[20:09:31.341 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[20:09:31.341 setWifiSTA] Wifi Station IP from DHCP
[20:09:31.374 onWiFiEvent] WiFi Station disconnected
[20:09:31.375 WARN sendChunks] Failed to send /data/MJPEG2SD.htm to browser 
[20:09:31.965 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
..........[20:09:37.450 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[20:09:37.451 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[20:09:41.075 startWifi] Wifi stats for josef_wifi - signal strength: -48 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[20:09:41.081 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect

```
and no, SD works just fine. tested speed too. will write it on other post.

josef2600, 2025-01-13:
i cant fined any setting to load HTML files from LittleFS. it only loads files from SD card. i only found  `#ifdef _LITTLEFS_H_` and nothing else.

josef2600, 2025-01-13:
just to be more clear, it is not just 1 file that fails. it is random files that fail to send.
```
[20:45:36.547 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[20:45:36.547 setWifiSTA] Wifi Station IP from DHCP
[20:45:36.622 onWiFiEvent] WiFi Station disconnected
[20:45:36.622 WARN sendChunks] Failed to send /data/common.js to browser 
[20:45:37.490 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
..........[20:45:42.649 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[20:45:42.651 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[20:45:46.275 startWifi] Wifi stats for josef_wifi - signal strength: -36 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[20:45:46.279 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect
[20:46:06.547 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[20:46:06.547 setWifiSTA] Wifi Station IP from DHCP
[20:46:06.581 onWiFiEvent] WiFi Station disconnected
[20:46:07.560 onWiFiEvent] WiFi Station connection to josef_wifi, using hostname: ESP-CAM_MJPEG_984B014AEC24
..........[20:46:12.657 WARN startWifi] SSID josef_wifi not connected WL_IDLE_STATUS 
[20:46:12.659 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
[20:46:16.073 startWifi] Wifi stats for josef_wifi - signal strength: -35 dBm; Encryption: WPA_WPA2_PSK; channel: 11
[20:46:16.260 onWiFiEvent] Wifi Station IP, use 'http://192.168.1.8' to connect

```

s60sc, 2025-01-13:
Either poor network connection quality or your router doesnt support pings, in which case switch off pings under wifi settings

josef2600, 2025-01-13:
but it say cant send file. is it the router problem?

s60sc, 2025-01-13:
failed to complete sending file to browser as wifi connection dropped

josef2600, 2025-01-13:
so, tomorrow i am going to business trip for few days. i will continue work after that.

closed 2025-01-13 by josef2600