s60sc/ESP32-CAM_MJPEG2SD #543: The camera only displays pink

PeterHonor, 2025-01-17:
I built with the latest version and then the camera just showed a pink color. With version 10.4.2 and Arduino Core 3.1.0 everything works fine. 

[10:11:12.650 logSetup] =============== ESP-CAM_MJPEG 10.5.2 ===============
[10:11:12.652 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[10:11:12.652 boardInfo] Flash 16.0MB, mode QIO @ 80Mhz
[10:11:12.652 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[10:11:12.653 logSetup] Compiled with arduino-esp32 v3.1.1
[10:11:12.653 printResetReason] Software reset via esp_restart
[10:11:12.653 printWakeupReason] Wakeup by reset
[10:11:12.653 setup] Selected board CAMERA_MODEL_ESP32S3_EYE_FREENOVE
[10:11:12.715 infoSD] SD card type SDHC, Size: 7.4GB, using 1 bit mode @ 40MHz
[10:11:12.716 listFolder] Sketch size 1.7MB
[10:11:12.874 listFolder] File: /data/configs.txt, size: 7417 bytes
[10:11:12.877 listFolder] File: /data/common.js, size: 46192 bytes
[10:11:12.879 listFolder] File: /data/MJPEG2SD.htm, size: 97KB
[10:11:12.880 listFolder] SD_MMC: 1.4MB used of 7.4GB
[10:11:12.880 loadConfig] Load config
[10:11:12.991 WARN checkConfigFile] Delete old /data/configs.txt
[10:11:12.995 deleteFolderOrFile] Deleting : /data
[10:11:12.995 deleteFolderOrFile] Folder /data contents
[10:11:12.998 deleteFolderOrFile] FILE : /data/configs.txt Size : 7417 bytes deleted
[10:11:13.023 deleteFolderOrFile] FILE : /data/common.js Size : 46192 bytes deleted
[10:11:13.043 deleteFolderOrFile] FILE : /data/MJPEG2SD.htm Size : 97KB deleted
[10:11:13.056 deleteFolderOrFile] Folder /data deleted
[10:11:13.179 checkConfigFile] Created /data/configs.txt from local store
[10:11:13.293 updateAppStatus] Enabling motion detection
[10:11:14.042 prepCam] Camera init OK for OV5640
[10:11:14.421 setWifiSTA] Wifi Station IP from DHCP
.[10:11:14.577 onWiFiEvent] WiFi Station connection to PeterHonor, using hostname: ESP-CAM_MJPEG_A4E3524CDADC
[10:11:14.834 onWiFiEvent] Wifi Station IP, use 'http://10.0.0.16' to connect
[10:11:14.946 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_A4E3524CDADC started, use 'http://192.168.4.1' to connect
[10:11:14.948 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_A.local
[10:11:19.563 startWifi] Wifi stats for PeterHonor - signal strength: -34 dBm; Encryption: WPA_WPA2_PSK; channel: 9
[10:11:19.564 startPing] Started ping monitoring - On
[10:11:19.567 startWebServer] Starting web server on port: 80
[10:11:19.568 startWebServer] Remote server certificates not checked
[10:11:19.568 startSustainTasks] Started 4 sustain tasks
[10:11:19.568 prepUpload] File uploads will use FTP server
[10:11:19.571 setupLamp] Setup WS2812 Lamp Led on pin 48
[10:11:19.571 WARN prepTemperature] No DS18B20 pin defined, using chip sensor if present
[10:11:19.572 WARN prepAudio] Microphone pins not defined
[10:11:19.572 WARN prepAudio] Amplifier pins not defined
[10:11:19.572 prepTelegram] Telegram not being used
[10:11:19.575 resetWatchDog] WatchDog started using task: ping
[10:11:19.576 getLocalNTP] Using NTP server: pool.ntp.org
[10:11:19.577 showLocalTime] Got current time from NTP: 17/01/2025 10:11:19 with tz: GMT0
[10:11:19.601 updateAppStatus] Enabling motion detection
[10:11:19.673 prepRecording] To record new AVI, do one of:
[10:11:19.673 prepRecording] - press Start Recording on web page
[10:11:19.673 prepRecording] - move in front of camera

[10:11:19.673 prepRecording] Camera model OV5640 ready @ 20MHz
[10:11:19.673 prepI2C] I2C bus shared with camera
[10:11:19.674 prepI2C] I2C initialised at 100kHz using pins SDA: 4, SCL: 5
[10:11:19.674 scanI2C] I2C device scanning
[10:11:19.683 scanI2C] I2C device OV5640/SSD1306 present at address: 0x3c
[10:11:19.692 scanI2C] I2C devices found: 1
[10:11:19.715 WARN setupBMx] BMx280 not available
[10:11:19.716 WARN setupRTC] DS3231 RTC not available
[10:11:19.717 prepRTSP] RTSP server started successfully with transports: Video & Subtitles
[10:11:19.718 prepRTSP] Connect to: rtsp://10.0.0.16:554
[10:11:19.718 checkMemory] Setup Free: heap 72744, block: 31732, min: 72660, pSRAM 2730760
[10:11:19.719 loop] =============== Total tasks: 21 ===============

[10:11:19.947 wgetFile] Downloading /data/common.js from /s60sc/ESP32-CAM_MJPEG2SD/master/data/common.js
[10:11:20.603 wgetFile] Downloaded /data/common.js, size 46192 bytes
[10:12:19.998 wgetFile] Downloading /data/MJPEG2SD.htm from /s60sc/ESP32-CAM_MJPEG2SD/master/data/MJPEG2SD.htm
[10:12:21.123 wgetFile] Downloaded /data/MJPEG2SD.htm, size 97KB
[10:12:34.782 wsHandler] Websocket connection: 55
[10:12:59.299 showStream] MJPEG: 132 frames, total 1.4MB in 6.6s @ 19.9fps
[10:13:44.411 showStream] MJPEG: 492 frames, total 5.2MB in 32.9s @ 14.9fps
[10:14:28.658 showStream] MJPEG: 232 frames, total 2.4MB in 25.8s @ 9.0fps
[10:25:01.236 WARN updateStatus] Trying to config gainceiling but feature not included
[10:25:13.664 WARN updateStatus] Trying to config ae_level but feature not included
[10:25:14.686 WARN updateStatus] Trying to config ae_level but feature not included
[10:25:19.521 showStream] MJPEG: 965 frames, total 10.1MB in 79.7s @ 12.1fps

[image]

s60sc, 2025-01-18:
hardware or configuration issue. impossible to diagnose unless someone else reports the problem

josef2600, 2025-01-19:
no problem with me. it works. OV5640

PeterHonor, 2025-01-20:
If I #define INCLUDE_I2C false then everything is ok. With version 10.4.2 and arduino-esp core 3.1.0 I don't have this problem. I think this is an I2C problem in arduino-esp core 3.1.1

josef2600, 2025-01-20:
yes i heard it too, but i think [s60sc](https://github.com/s60sc) has fixed it. it has been more than a week since the fix. you should download it.

PeterHonor, 2025-01-21:
@josef2600 : **I just downloaded latest version 10.5.3. Nothing changes (I have nothing connected to my board)**

_[12:05:06.793 logSetup] =============== ESP-CAM_MJPEG 10.5.3 ===============
[12:05:06.795 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[12:05:06.795 boardInfo] Flash 16.0MB, mode QIO @ 80Mhz
[12:05:06.796 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[12:05:06.796 logSetup] Compiled with arduino-esp32 v3.1.1_

**If I replace ov5640 with ov2640 everything works fine. Then I discovered the problem is that the I2C address of the ov5640 is the same as the address of the SSD1306. If i change I2C address of SSD1306 from 0x3C to 0x3D (in periphsI2C.cpp) then I can see images from the camera, but I can't change any camera parameters. When I change I get a warning: feature not included:**

_[10:25:01.236 WARN updateStatus] Trying to config gainceiling but feature not included
[10:25:13.664 WARN updateStatus] Trying to config ae_level but feature not included
[10:25:14.686 WARN updateStatus] Trying to config ae_level but feature not included_

**In the end I had to #define INCLUDE_I2C false and everything is OK now, but no I2C**

josef2600, 2025-01-21:
when somebody say `if you have anything connected to the board` ,it doesn't mean just physically! mainly it means in the code.
on any code or board, you have to check the hardware and know what is connected to what. specially when you have connected a special device to your board, like a camera, never use the same i2c (or hardware) for another thing, like whatever you are using, to it. unless you have full knowledge of what you are doing and what the code is doing. for example, this code uses LEDC channel 1 and timer 1 for creating camera clock, but most other codes, use channel 0 and timer 0. i chose 1 because of things like this. everything and everyone by default use channel 0, and then everything gets messy. 
for your particular problem, you have to use i2c channel 1 and create a new driver for your i2c. DO NOT USE the camera i2c or pins. i hope i was clear about what is wrong this time. besides, i don't think you have a direct access to its code and you are creating your own driver on top of the camera driver and that destroys everything!

s60sc, 2025-01-21:
`[10:25:01.236 WARN updateStatus] Trying to config gainceiling but feature not included`
I'll look into this.

I2C sharing was meant for ESP32, on the ESP32S3 you have spare pins so assign separate I2C pins so you dont have to share.

s60sc, 2025-01-21:
issue [#718](https://github.com/espressif/esp32-camera/issues/718) raised on esp camera library

PeterHonor, 2025-01-22:
@s60sc @josef2600 : Yeah! I changed _Edit Configs->Peripherals->I2C SDA pin if unshared_ from -1 to another pin and everything is OK now. Thanks for your help and support

closed 2025-01-22 by s60sc