s60sc/ESP32-CAM_MJPEG2SD #551: Uart compilation error

GramOfCarbs, 2025-01-19:
Hey all, quick question. I got this error here: exit status 1              Compilation error: invalid conversion from 'int' to 'uart_port_t' [-fpermissive]
it took me to the uart.cpp tab. I originally fond a tutorial for this server on Max Imagination's channel, and I asked about this error and he sent me here. Any idea as to why this error occurred?

s60sc, 2025-01-20:
download latest version

GramOfCarbs, 2025-01-20:
> download latest version

I got the same error.

emc02, 2025-01-29:
had the same error, changed line 44 in uart.cpp

static int uartId; to static uart_port_t uartId;

workes perfect now.

s60sc, 2025-01-29:
Using old version, download latest version

emc02, 2025-01-29:
with new version it's working fine, but that's the fix for the latest release ;-)

closed 2025-02-06 by s60sc