s60sc/ESP32-CAM_MJPEG2SD #566: Please let ntp server endpoint be configurable

alithechemist, 2025-02-01:
As per best practice ad common sense, I use a second stub WIFI for IOT devices in order to prevent them from connecting to the internet, at all.

If I want o connect to these devices from my main WIFI i have an ad-hoc gateway and routes.

I do this in order to avoid gadgets, cameras and other WIFI enabled devices from sending information about my setup all over the internet, increasing attack surface, track-ability and profiling.

Think just about my ISP, who is selling my raw traffic to data brokers (your ISP too!): even the transmissions are encrypted, endpoints and payload size can reveal what devices i have on my network which in turn can reveal a lot about me, and I don't want that. This information can be subsequently used against me, to do things that range from identity theft to scams, intrusion etc.

So first i had to work around the data folder download problem: I just created the folder and copied the files in it.

I only have this problem with NTP ... 

```
[00:12:07.914 WARN getLocalNTP] Not yet synced with NTP 
[00:12:37.910 getLocalNTP] Using NTP server: pool.ntp.org
```

I have my infrastructure with its own NTP server just for the reasons I described before.
I'd like to ask you to make NTP server configurable with a default, so people who run their own infrastructure can take advantage of this.

Obviously if you are open to this I can take care of a pull requests, though I think you know by heart where to change things.

Thank you in advance,
Chip

alithechemist, 2025-02-01:
I replaced the ntp server endpoint in the configuration but now I'm experiencing another problem:

```
[11:23:40.524 WARN remoteServerConnect] Timed out connecting to server: api.ipify.org, Err: -1, ERROR - Generic error 
```

Why does the application need to know my external ip address?

Thanks in advance
Chip

s60sc, 2025-02-01:
Edit Config / Wifi

[image]

alithechemist, 2025-02-01:
Thanks! So turns out i failed the installation, and by seeing those logs entries i thought they were preventing the installation from completing.

My mistake, sorry, I see now the options.

Thanks,
Chip

closed 2025-02-01 by alithechemist