s60sc/ESP32-CAM_MJPEG2SD #567: Not an issue, an LED request

Stewie-040882, 2025-02-02:
could the frequency of the led pin be modified to around 400hz please, ive tried manually changing it, but cant figure it out.. the current frequency for the ai thinker gives a high frequency whine. Thanks

s60sc, 2025-02-02:
line 354 in `peripherals.cpp`, change `5000` to `400`

`    ledcAttach(lampPin, 5000, DUTY_BIT_DEPTH); // freq, resolution`

Stewie-040882, 2025-02-05:
thats what id done, but since i didnt refresh the browser it didnt refresh the modification (sorry).. is it possible to add the led brightness under RC Config, or add a brightness slider while streaming, i only ask as when i toggle the built in LED on, it gets crazy hot. Thanks

closed 2025-03-14 by s60sc