s60sc/ESP32-CAM_MJPEG2SD #571: Startup Failure: Insufficient PSRAM for app: 2.0MB in Esp32-Cam Board

JenulLithmal2025, 2025-02-05:
When I hit upload button, it's uploading with no error but it's showing a error in the **Serial Monitor** called 

logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.257 logSetup] =============== ESP-CAM_MJPEG 10.5.4 ===============
[00:00:00.267 boardInfo] Chip ESP32-D0WD-V3, 2 cores @ 240Mhz, rev 3
[00:00:00.267 boardInfo] Flash 4.0MB, mode QIO @ 80Mhz
[00:00:00.278 boardInfo] PSRAM 2.0MB, mode QSPI @ 80Mhz
[00:00:00.278 logSetup] Compiled with arduino-esp32 v3.1.1
[00:00:00.289 printResetReason] Power on reset
[00:00:00.289 printWakeupReason] Wakeup by reset
[00:00:00.289 setup] Selected board CAMERA_MODEL_AI_THINKER
[00:00:00.651 infoSD] SD card type SDHC, Size: 31.2GB, using 1 bit mode @ 40MHz
[00:00:00.651 listFolder] Sketch size 1.5MB
[00:00:00.977 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[00:00:00.980 listFolder] File: /data/common.js, size: 46192 bytes
[00:00:00.983 listFolder] File: /data/MJPEG2SD.htm, size: 97KB
[00:00:00.995 listFolder] File: /data/configs.txt, size: 7819 bytes
[00:00:00.996 listFolder] SD_MMC: 448KB used of 31.2GB
[00:00:00.996 loadConfig] Load config
[00:00:01.288 updateAppStatus] Enabling motion detection
[00:00:01.487 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.507 WARN startWifi] SSID not connected not available, use AP
[00:00:01.509 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_B0F3C0182EC8 started, use 'http://192.168.4.1' to connect
[00:00:04.523 startPing] Started ping monitoring - On
[00:00:04.527 startWebServer] Starting web server on port: 80
[00:00:04.528 startWebServer] Remote server certificates not checked
[00:00:04.538 WARN setup] Startup Failure: Insufficient PSRAM for app: 2.0MB
[00:00:04.538 loop] =============== Total tasks: 13 ===============

[00:00:04.613 onWiFiEvent] WiFi AP client connection
[00:00:08.715 WARN sendChunks] Failed to send /data/MJPEG2SD.htm to browser
[00:00:09.525 resetWatchDog] WatchDog started using task: ping

I upload this with esp32 Base board not from FTDI Adaptor. I follow the video from max imagination - https://www.youtube.com/watch?v=k_PJLkfqDuI, how can solve this problem, I trying esp32-cam board for the first Time.

[image]
[image]

s60sc, 2025-02-06:
As per readme, need board with min 4MB psram

JenulLithmal2025, 2025-02-06:
Ohh, thank you so much your response🥹

Can I know this board has min 4mb PSRAM. 

[image]

[image]

rjsachse, 2025-02-07:
if that is a different board then test. otherwise it is saying 2mb psram.

closed 2025-03-14 by s60sc