s60sc/ESP32-CAM_MJPEG2SD #573: Troubleshooting

giampierobartolomei, 2025-02-07:
### Discussed in https://github.com/s60sc/ESP32-CAM_MJPEG2SD/discussions/572

<div type='discussions-op-text'>

<sup>Originally posted by **giampierobartolomei** February  7, 2025</sup>
Hi, thank you for this very useful repository. I successfully installed the firmware and used the web application integrating it with bt functionalities (I didn't modify the original code, but only added a library written by me). Now i am encountering some problems:
1) after reuploading the code sometimes my serial monitor is empty, and every time i upload i need to select again the port beacuse the device disconnects. When the serial monitor is empty the device can anyway comunicate with BT-
2) My Wifi network failed the ping test and i can't connect to the AP in order to modify my setting and try a new Wi-Fi connection. I tried to delete /data folder from my SD but it seems that my Wifi keeps saved in some sort of cache (sorry i'm not so expert).
3) Is there a way to start the device every time in AP mode without Wifi connection? my AP doesn't appear in my wifi list as before.
4)Extra question, is there a way to disable motion detect by default?

Serial monitor output:
[00:00:10.382 loadConfig] Load config
[00:00:10.496 checkConfigFile] Created /data/configs.txt from local store
[00:00:10.607 updateAppStatus] Enabling motion detection
[00:00:10.785 prepCam] Camera init OK for OV2640
[00:00:11.121 setWifiSTA] Wifi Station IP from DHCP
...[00:00:12.219 onWiFiEvent] WiFi Station disconnected
..[00:00:13.280 onWiFiEvent] WiFi Station disconnected
.....[00:00:16.144 WARN startWifi] SSID ISTC_Floor_4 not connected unable to connect 
[00:00:16.146 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_0C338E188534 started, use 'http://192.168.4.1' to connect
[00:00:16.149 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_0.local
[00:00:19.362 startWifi] Wifi stats for ISTC_Floor_4 - signal strength: -72 dBm; Encryption: WPA2_PSK; channel: 6
[00:00:19.363 startWifi] Wifi stats for ISTC_Floor_4 - signal strength: -73 dBm; Encryption: WPA2_PSK; channel: 6
[00:00:19.364 startPing] Started ping monitoring - On
[00:00:19.366 startWebServer] Starting web server on port: 80
[00:00:19.367 startWebServer] Remote server certificates not checked
[00:00:19.367 startSustainTasks] Started 1 sustain tasks
[00:00:19.368 setupLamp] Setup PWM Lamp Led on pin 21
[00:00:19.370 prepAudio] Sound capture is available using PDM mic on I2S0 with gain 0
[00:00:19.370 WARN prepAudio] Amplifier pins not defined 
[00:00:19.392 updateAppStatus] Enabling motion detection
[00:00:19.503 prepRecording] To record new AVI, do one of:
[00:00:19.503 prepRecording] - press Start Recording on web page
[00:00:19.504 prepRecording] - move in front of camera
 
[00:00:19.504 prepRecording] Camera model OV2640 ready @ 20MHz
[00:00:19.504 checkMemory] Setup Free: heap 84436, block: 31732, min: 84412, pSRAM 5653880
Characteristic defined!
[00:00:19.819 loop] =============== Total tasks: 25 ===============

[00:00:24.365 resetWatchDog] WatchDog started using task: ping
[00:00:24.365 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[00:00:24.365 setWifiSTA] Wifi Station IP from DHCP
...[00:00:25.417 onWiFiEvent] WiFi Station disconnected
.......[00:00:29.368 WARN startWifi] SSID ISTC_Floor_4 not connected unable to connect 
[00:00:29.370 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_0.local
[00:00:38.143 startWifi] Wifi stats for ISTC_Floor_4 - signal strength: -69 dBm; Encryption: WPA2_PSK; channel: 6
[00:00:38.144 startWifi] Wifi stats for ISTC_Floor_4 - signal strength: -70 dBm; Encryption: WPA2_PSK; channel: 6

I also tried to clone the repository(coping the /data folder of the repo on microSD) and run the 'clean' code but the mentioned problems persists.</div>

s60sc, 2025-02-07:
i've not used bt, but I dont think it can coexist with wifi as there is only one radio

giampierobartolomei, 2025-02-08:
Hi, as I said the problem persist when I run the Original code from your repository (so without bluetooth)

s60sc, 2025-02-08:
your wifi signal strength is low, try moving esp closer to router.
To stop ping: `Edit Config / Wifi / Use ping` then press `Save `and `Reboot`

giampierobartolomei, 2025-02-08:
Ok thanks, last question.
Even if the LOG INF prints confirm the AP connection, i can't find it in Wi-Fi list on my devices, so i can't modify my preferences from the webapp. Then, I don't know how, even if i delete files in /data folder in my SD, my Wi-Fi "ISTC_Floor_4" SSID & pass keeps saved, so i can't restart the AP like in first installation. So, i want to ask to you how to set the CAM to work in AP mode by putting the correct settings by modifing the code, in way to have a stable AP connection with my XIAO ESP 32S3. For my application it would also perfect to work only in AP mode.

s60sc, 2025-02-08:
In `Access Settings / SSID` delete entry then `Save Settings` then `Reboot`
ESP AP only has limited range, devices need to be close by

closed 2025-03-14 by s60sc