s60sc/ESP32-CAM_MJPEG2SD #575: Wifi Password URL encoding stripping some characters

moosemanca, 2025-02-18:
I will preface this with I am EXTREMELY new to this project and programming for ESP32. I was struggling with getting this going and thought I was just a dumbdumb, but then I noticed something and after some research, I went back to this and noticed an oddity in the initial setup page.

When passing in dangerous characters to the initial setup page, the URL Encoding causes the password to go in the URL blank

The following:

[image]

Yields this result in chrome:

[image]

If it isn't too complex, perhaps the endpoint could return yuck (400?) but that might not be simple. I do not know.

Hammad-66, 2025-02-18:
same issue

s60sc, 2025-02-18:
see issue [263](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/263)

closed 2025-03-14 by s60sc