s60sc/ESP32-CAM_MJPEG2SD #578: Startup Failure: Insufficient PSRAM for app: 2.0MB in Esp32-Cam Board

MichailKantza, 2025-02-21:
Hello(and sorry that Im writing this here, but I didn't know were else to send it ),

I'm working on a drone project where I want my ESP32-CAM to:

-    Provide a live stream.
-    Record video when the drone is armed and stop recording when disarmed. The video should be saved as an AVI file on the microSD card inserted into the ESP32-CAM.

To achieve this, I’ve been using this code and followed the instructions from this [video tutorial](https://www.youtube.com/watch?v=k_PJLkfqDuI&t=470s).
Problem with Latest Code Version:

When I use the latest code version from the GitHub repository, the upload works fine, but when I try to access the stream via the IP address, I get the following error message:

Startup Failure: Insufficient PSRAM for app: 2.0MB

Workaround with Older Versions:

I tried installing earlier versions of the code and the board, because I saw in the video that the features that I want work:

ESP32-CAM_MJPEG2SD version 8.4a from GitHub.
ESP32 board version 2.0.5 in the Arduino IDE.

With this setup, the system works but with some issues. The main problem is that the ESP32-CAM access point becomes unstable, and I can't stay connected for long. Despite this, I managed to stream video and save recorded AVI files on the SD card.

Hardware Setup:
I’m using the ESP32-CAM module with the ESP32-CAM-MB adapter, connected to my flight controller (SpeedyBee F405 V4) as follows:

5V (SpeedyBee F405 V4) → 5V (ESP32-CAM module)
G (SpeedyBee F405 V4) → GND (ESP32-CAM module)
R3 (SpeedyBee F405 V4) → U0T (ESP32-CAM module)
T3 (SpeedyBee F405 V4) → U0R (ESP32-CAM module)

My Questions:

-    Can I achieve stable performance with the older version of the GitHub code and the ESP32 board version 2.0.5?
-   Is there any way to fix the "Insufficient PSRAM" issue with the latest code version?
-    How can I make the ESP32-CAM's access point more stable for longer connections?

Is there any way to fix the "Insufficient PSRAM" issue with the latest code version?
How can I make the ESP32-CAM's access point more stable for longer connections?

s60sc, 2025-02-21:
The app does not work with fake boards with only 2MB PSRAM, buy a genuine board with 4MB PSRAM.

MichailKantza, 2025-02-22:
Okay I going to do that. Thank you so much for the help!

closed 2025-02-22 by MichailKantza