s60sc/ESP32-CAM_MJPEG2SD #585: Camera error 0x105

raptooor15, 2025-03-01:
Im not sure what this error means: Startup Failure: Camera init error 0x105:ESP_ERR_NOT_FOUND on CAMERA_MODEL_ESP32_CAM_BOARD
Im using ESP32-CAM (also called AI thinker), othervise running unedited code (apart from board definition and SD card = 0). Also, when visiting the ip adress, only the same error shows up along with *check log* and *Reboot ESP after fix*

[00:00:00.487 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.487 logSetup] =============== ESP-CAM_MJPEG 10.5.4 ===============
[00:00:00.498 boardInfo] Chip ESP32-D0WDQ6, 2 cores @ 240Mhz, rev 1
[00:00:00.498 boardInfo] Flash 4.0MB, mode QIO @ 80Mhz
[00:00:00.508 boardInfo] PSRAM 4.0MB, mode QSPI @ 80Mhz
[00:00:00.509 logSetup] Compiled with arduino-esp32 v3.1.1
[00:00:00.519 printResetReason] Power on reset
[00:00:00.519 printWakeupReason] Wakeup by reset
[00:00:00.519 setup] Selected board CAMERA_MODEL_ESP32_CAM_BOARD
[00:00:00.532 listFolder] Sketch size 1.6MB
[00:00:00.864 listFolder] File: /data/configs.txt, size: 7819 bytes
[00:00:00.866 listFolder] LittleFS: 24576 bytes used of 896KB
[00:00:00.868 loadConfig] Load config
[00:00:01.146 updateAppStatus] Enabling motion detection
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8

[00:00:02.673 logSetup] Setup RAM based log, size 7168, starting from 819

[00:00:02.674 logSetup] =============== ESP-CAM_MJPEG 10.5.4 ===============
[00:00:02.684 WARN initBrownout] Brownout occurred due to inadequate power supply 
[00:00:02.684 boardInfo] Chip ESP32-D0WDQ6, 2 cores @ 240Mhz, rev 1
[00:00:02.694 boardInfo] Flash 4.0MB, mode QIO @ 80Mhz
[00:00:02.695 boardInfo] PSRAM 4.0MB, mode QSPI @ 80Mhz
[00:00:02.705 logSetup] Compiled with arduino-esp32 v3.1.1
[00:00:02.705 printResetReason] Software reset due to brownout
[00:00:02.716 printWakeupReason] Wakeup by reset
[00:00:02.716 setup] Selected board CAMERA_MODEL_ESP32_CAM_BOARD
[00:00:02.729 listFolder] Sketch size 1.6MB
[00:00:03.061 listFolder] File: /data/configs.txt, size: 7819 bytes
[00:00:03.063 listFolder] LittleFS: 24576 bytes used of 896KB
[00:00:03.065 loadConfig] Load config
[00:00:03.343 updateAppStatus] Enabling motion detection
[00:00:04.053 setWifiSTA] Wifi Station IP from DHCP
[00:00:04.076 WARN startWifi] SSID  not connected not available, use AP 
[00:00:04.078 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_44E2AEC55494 started, use 'http://192.168.4.1' to connect
[00:00:07.692 startPing] Started ping monitoring - On
[00:00:07.696 startWebServer] Starting web server on port: 80
[00:00:07.697 startWebServer] Remote server certificates not checked
**[00:00:07.707 WARN setup] Startup Failure: Camera init error 0x105:ESP_ERR_NOT_FOUND on CAMERA_MODEL_ESP32_CAM_BOARD** 
[00:00:07.717 loop] =============== Total tasks: 13 ===============

[00:00:12.693 resetWatchDog] WatchDog started using task: ping

s60sc, 2025-03-01:
[00:00:02.684 WARN initBrownout] Brownout occurred due to inadequate power supply

raptooor15, 2025-03-01:
Had it on normal USB cable. Still i dont think that is the issue really

closed 2025-03-14 by s60sc