s60sc/ESP32-CAM_MJPEG2SD #586: Startup Failure: Check SD card inserted

6stitch9, 2025-03-03:
I'm using Aideepen Esp32 which is supposedly a AI_THINKER clone but every time I upload the code I get the "Startup Failure: Check SD card inserted" error. I've looked at every other case with this problem and have 1) commented out all but my model 2)switched sd cards 3) Set allocation unit size to 8192. These are all the fixes I've seen but the problem persists, any ideas to fix it?

[image]

Then once I connect to the network it gives the following error. And on the front end all you can see if a blank page that says; Setup Failure: Check SD card inserted, "check log", and "Reboot ESP after fix"

[image]

s60sc, 2025-03-03:
hardware issue - check pinout, check sd reader works with an example sketch

6stitch9, 2025-03-08:
SD card reader fails with example sketch, is there anyway to go about fixing this or do I need to purchase new esp32-cam module. If so what model should I go for for the best performance and least hassle.

closed 2025-03-14 by s60sc