s60sc/ESP32-CAM_MJPEG2SD #587: OV5640 Record issue

rjsachse, 2025-03-08:
I bought a ov5640 to test and noticed a issue with recording frame size I have it set to 640x480 (VGA) when played in VLC it shows it 640x960 so for some reason it doubles the height. I have not looked into this, or tried OV2640

josef2600, 2025-03-23:
hi @rjsachse ! currently i am busy on something else, But, the ov5640 driver is very bad! it is not really the espressif part, although they wrote it, but there is no real basic driver or datasheet for it to write a good driver. it is full of inconsistency and very problematic registering in it. in your specific part, in short, the camera does not support that resolution. it has to crop it from a larger size and it can not produce that particular resolution. if i remember correctly, it only supports 5 different resolution. it is in the datasheet. i was working on it for a long time, but i got busy with my real job and didn't have the the to finish a driver. and i didn't got any one to help too. and that is not all! the original driver is from IDF and it wont work with Arduino directly. but if you want to work on it, here it is:
https://github.com/espressif/esp32-camera
also, if you do want to work with it, it has 2 big problem. clocking, and "ratio_settings_t ratio_table" in `ov5640_settings.h`. they are all wrong! i did manage to fiddle with the clock, but didn't finish it, since it has so many parts and there is zero info about them. but most important thing is the ratio table.
any way, good luck.

rjsachse, 2025-03-24:
Thanks mate, I know about time. Being working on a doorbell camera with 2 way audio I finished the rtsp server and 80% on Onvif server and just done a port of SpeexDSP so don't have much time either till these are finished. I was testing a ov5640 camera just for benchmarking bit yer seems to have a few issues so I'll stick with ov2640 for now. After finishing the software then I need to finish the hardware design. Thanks for the info 👍

josef2600, 2025-03-24:
thanks. the main problem is the lack of information. i think they where made specifically for some companies and the real datasheet is protected by NDA. and also these cameras are way old and no company is making them for a long time, so no new data for them. on the other side, the esp32 doesn't support DVP or serial hardware for the new cameras or better say, no camera! since they all are serial and not parallel. also, i think your library about streaming is used in this project, thank you for that.

rjsachse, 2025-03-25:
I see the ESP32-P4 has MIPI Camera Serial Interface (CSI) and ISP Digital Video Port (ISP DVP) so hopefully when S4 comes out there would be better camera support

josef2600, 2025-03-25:
but that causes many problems! ESP32-P4 is not a regular ESP32, instead it is a made up copy of of the copy of other microcontrollers. and have you seen its prices? why the hell would i want to work with such an unstable chip? all it has and many many more already exist in ST or microchip and they don't have these problems. even their prices can be the same or lower. and do consider how much it cost to build. sure, PIC32 is more expensive, but it doesn't hang or has hardware problem or, specially, driver and code problems.
in short, i do not think it would be a good choice to do anything with it. so far, all ESP32's have been super chip with a lot of good stuff and specially onboard WIFI and BLT and a lot of SRAM and PSRAM and large amount of flash. but this thing has non!
and even if we don't care about those, there is not going to be any reliable libraries or codes for it for years! and even when it comes out, you know how many problems it is going to have.
for example, right now, it is a big problem for me right now! i am working on the ZigBee with ESP32-C6. the chip was out in 2022, but its ZigBee lib's still unfinished! and the documents are so confusing that i haven't understood them fully, or even 10% of them. 
i could have work with many other chips too, but i thought it was easy to work with it and have a WIFI to show off! how wrong i was!

closed 2025-03-25 by rjsachse