s60sc/ESP32-CAM_MJPEG2SD #591: ESPS3 wroom-1 AP does not show up - log seems clear, SD card has been formatted

averter, 2025-03-12:
I was adding a microphone to my esp32s3 in [a separate issue](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/589) when started having connectivity issues: it doesn't connect to the wifi and there is no AP to connect to (the unit is not accessible in any way). The log is shown below (seems clear to me). I'm also showing the changes to the software (ssid and pass obfuscated for obvious reasons) 

[image]

````
SP-CAM_MJPEG_8.local
1.100Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac

[00:00:00.050 logSetuSP-CAM_MJPEG_8.local
[00:00:04.122 startPing] Started ping monitoring - On
[00:00:04.124 startWebServer] Starting web server on port: 80
[00:00:04.125 startWebServer] Remote server certificates not checked
[00:00:04.135 startSustainTasks] Started 1 sustain tasks
[00:00:04.136 WARN prepAudio] Microphone pins not defined 
[00:00:04.146 WARN prepAudio] Amplifier pins not defined 
[00:00:04.166 updateAppStatus] Enabling motion detection
[00:00:04.274 prepRecording] To record new AVI, do one of:
[00:00:04.274 prepRecording] - press Start Recording on web page
[00:00:04.274 prepRecording] - move in front of camera
 
[00:00:04.285 prepRecording] Camera model OV2640 ready @ 20MHz
[00:00:04.285 checkMemory] Setup Free: heap 156768, block: 94196, min: 156760, pSRAM 5652120
[00:00:04.296 loop] =============== Total tasks: 18 ===============

[00:00:09.124 resetWatchDog] WatchDog started using task: ping
````

s60sc, 2025-03-12:
dont hard code values they get overwritten, thats why the comment says
`/** Use the web interface to configure wifi settings **/`

averter, 2025-03-12:
I'm baffled...I've reset those lines to default, i.e. they are now blank, flashed the software to the ESP32S3 and not only it does not show the AP it somehow remembers and tries to connect to my old network...I've also tried to format the sd card and the behavior stays the same...what the hell is happening?

averter, 2025-03-12:
Something really strange is happening. I've managed to reconnect with the ESP32S3, then I have connected the pins to get audio (I've used ones without symbols as discussed in the other thread, 15, 16 and 17) and then I've lost connection again...
So for some reason when the mic is connected, I declare the pins via the web interface, save and reset it messes up the wifi connection. Here is the log after the save and reset.
````
[15:54:36.526 setWifiSTA] Wifi Station IP from DHCP
...[15:54:37.638 onWiFiEvent] WiFi Station disconnected
.......[15:54:41.528 WARN startWifi] SSID thenameof my_network not connected unable to connect 
[15:54:41.529 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_8.local
[15:54:45.353 startWifi] Wifi stats for thenameof my_network - signal strength: -40 dBm; Encryption: WPA2_PSK; channel: 1
[15:55:06.514 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[15:55:06.514 setWifiSTA] Wifi Station IP from DHCP
...[15:55:07.545 onWiFiEvent] WiFi Station disconnected
........[15:55:12.016 WARN startWifi] SSID thenameof my_network not connected unable to connect 
[15:55:12.017 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_8.local
[15:55:15.640 startWifi] Wifi stats for thenameof my_network - signal strength: -38 dBm; Encryption: WPA2_PSK; channel: 1
[15:55:36.514 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[15:55:36.514 setWifiSTA] Wifi Station IP from DHCP
...[15:55:37.583 onWiFiEvent] WiFi Station disconnected
........[15:55:42.016 WARN startWifi] SSID thenameof my_network not connected unable to connect 
[15:55:42.017 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_8.local
[15:55:45.641 startWifi] Wifi stats for thenameof my_network - signal strength: -40 dBm; Encryption: WPA2_PSK; channel: 1
````

closed 2025-03-12 by s60sc