s60sc/ESP32-CAM_MJPEG2SD #592: wifI ISSUE

frabar55, 2025-03-13:
Hi Team,
I installed the project correctly withou any error.
But when I inserted the wifi user/psw, the system do not connet to wifi with the followin error:
00:29:44.160 WARN pingTimeout] Failed to ping gateway, restart wifi ... 
[00:29:44.160 setWifiSTA] Wifi Station IP from DHCP
..[00:29:44.760 onWiFiEvent] WiFi Station disconnected
........[00:29:49.178 WARN startWifi] SSID Asus not connected WL_CONNECT_FAILED 
[00:29:49.179 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_3.local
[00:29:52.193 startWifi] Wifi stats for Asus - signal strength: -28 dBm; Encryption: Not listed; channel: 11
[00:29:52.194 startWifi] Wifi stats for Asus - signal strength: -71 dBm; Encryption: Not listed; channel: 11
[00:29:52.204 startWifi] Wifi stats for Asus - signal strength: -75 dBm; Encryption: Not listed; channel: 11

The environment is as follow:
ESP32-S3 CAM (eye mode)
arduino 3.1.3

Could you pleas help me?
Regards

frabar55, 2025-03-13:
Can I change the wifi param without doing again the firmware upload?

frabar55, 2025-03-14:
I checked the password and it is correct, the error continue

frabar55, 2025-03-14:
I've checked the ssd content:
- common.js, with 5 registration files
- configs.txt, with paramenters (a lot)
this is an exstract:
RTSP_Name~~8~T~RTSP Auth Username
RTSP_Pass~~8~T~RTSP Auth Password
ST_SSID~Asus~99~~na
ST_gw~~0~T~Router IP address
ST_ip~~0~T~Static IP address
ST_ns1~~0~T~DNS server
ST_ns2~~0~T~Alt DNS server
ST_sn~255.255.255.0~0~T~Router subnet

s60sc, 2025-03-14:
wifi problem, nothing to do with app

closed 2025-03-14 by s60sc

frabar55, 2025-03-14:
sorry but that seems like a non-answer, anyway sorry if I wasted your time, I will do without your project
greetings