s60sc/ESP32-CAM_MJPEG2SD #595: File does not exist or cannot be opened

Seazer-x, 2025-04-05:
File does not exist or cannot be opened: /data/MJPEG2SD.htm
FSFileImpl(): fopen(/sdcard/data/common.js) failed

[image]

Seazer-x, 2025-04-05:
> File does not exist or cannot be opened: /data/MJPEG2SD.htm FSFileImpl(): fopen(/sdcard/data/common.js) failed
> 
> [image]

[image]

s60sc, 2025-04-05:
missing files need to be installed manually if not automatically downloaded (no network connection)

Seazer-x, 2025-04-05:
[image]
how to upload to esp32s3?

Seazer-x, 2025-04-06:
[image]
how to upload them to esp32？

closed 2025-06-12 by s60sc