s60sc/ESP32-CAM_MJPEG2SD #613: HTML page not resonding with ESP32S3

MacGH23, 2025-05-20:
Hi,

I use a ESP32 normally and work without problems.
Now I got a Freenove ESP32S3 Cam (clone) which looks like working,
because motion detection is running und capturing videos to SD, but
the HTML page is not recognizing any input (expect OTA button)
I get only the the static page as on the on the SDcard.
Arduino setting should be OK, followed the instructions.

Any Hint ?

s60sc, 2025-05-20:
download latest `common.js` as it had a bug yesterday

MacGH23, 2025-05-20:
Unfortunately it doesn't help.
I setup competently new, format SD and upload new sketch

MacGH23, 2025-05-21:
I enabled debug

> 10:24:58.364 -> [08:24:57.618 WARN sendChunks] Low memory - failed to send to browser: /data/MJPEG2SD.htm, err ESP_ERR_HTTPD_RESP_SEND 
10:24:58.411 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
10:24:58.445 -> 
10:24:58.445 -> Core  1 register dump:
10:24:58.445 -> PC      : 0x40381cec  PS      : 0x00060333  A0      : 0x80383151  A1      : 0x3fcd0f60  
10:24:58.445 -> A2      : 0x3fccccd4  A3      : 0xffffffff  A4      : 0x00060520  A5      : 0x00060323  
10:24:58.445 -> A6      : 0x00ff0000  A7      : 0x0000abab  A8      : 0x00860022  A9      : 0x00000001  
10:24:58.445 -> A10     : 0x3fcccd00  A11     : 0x00000000  A12     : 0xb33fffff  A13     : 0x00060523  
10:24:58.479 -> A14     : 0x00000001  A15     : 0x0000abab  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
10:24:58.479 -> EXCVADDR: 0x00000005  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffe1  
10:24:58.479 -> 
10:24:58.479 -> 
10:24:58.479 -> Backtrace: 0x40381ce9:0x3fcd0f60 0x4038314e:0x3fcd0f80 0x420057c7:0x3fcd0fa0 0x420057ef:0x3fcd0fc0 0x42016074:0x3fcd0fe0 0x42016919:0x3fcd1000 0x42016b19:0x3fcd1050 0x4206f3d9:0x3fcd1070 0x4206e40c:0x3fcd10a0 0x4206ead4:0x3fcd1140 0x4206d740:0x3fcd1160 0x42118cce:0x3fcd1180 0x4206d8eb:0x3fcd11a0 0x40380fbe:0x3fcd1230
10:24:58.521 ->

closed 2025-06-12 by s60sc