s60sc/ESP32-CAM_MJPEG2SD #623: Upgraded ESP 32 Board camera

sinole1987, 2025-06-12:
Hello I want to thank you for the great software again. It makes great use of this small board, but I wasnt happy with picture quality I was getting out of OV2640 I was looking on ali for ov5640 when I found this [improved board](https://www.aliexpress.us/item/3256807708233463.html?spm=a2g0o.order_list.order_list_main.5.21ef1802JNJ8UR&gatewayAdapt=glo2usa). Do you know what changes if any I should make to software to use it on this board? it has more ram and can use bigger SD card and it comes out of box with better sensor. I know the AI thinker board should be modified to use with OV5440. I appriciate you input.

s60sc, 2025-06-12:
Not worth using ov5640 with ESP32, use more powerful ESP32-S3 board

giancarlodomanico, 2025-06-14:
> I think the board I linked above is not exactly S3 but it is close. I think you are talking about XIAO board which also has 8mb external ram. I screen shoted features of the board, if you are hesitant on opening the ali express link. I just want to know how to program this board to make most of it.
> 
> [image]

The board you linked cost like a ESP32-CAM S3, with same 8 MB RAM and OV5640 camera. Why use a less powerful MCU if you can buy at the same price a S3 ?

josef2600, 2025-06-27:
hi!
@sinole1987 -> first, this was never an issue. when you have a question, or need help with whatever, please put it in the discussion, not issue.
again, not an issue for this project, but i did feel the low camera quality images. the best answer i could com up with, was the original tabling problems in the main Driver (the pixel ratio). not this project.
@s60sc please add a section in the front page for "issues not from this project!". i am getting real tired of saying these. the driver for ov5640 has many problems that there are no real solve for them. since it is a very old camera and it is out of manufacturing and most important, it does NOT have a real user manual. at least half of its important registers dont exist on any datasheet. i even checked the Linux driver, it has the same issues, and i checked the most recent drivers from ST micro and microchip, the same issue but with added problems! like, they only do 480p max! not even 720p. probably in the next few weeks i will return to them, since i like this hardware combo.
and as the power management system to prevent camera from overheating, it doesn't need any hardware pins, it can be don with software easily. its just a bit to make it zero. i dont remember it, but it was in the datasheet. the point of it is to when we are not using the camera, like, when not streaming, we put the camera on power saving mode so it wont get hot. that is not a real solution for overheating, but a way to avoid it as best as we can do right now. the main problem is the setting of the camera that are wrong and the registers are not explained in the datasheet, unless you do the serial communicating instead of parallel.

sinole1987, 2025-07-02:
Man, I appologize for inconviniencing you. Although I don't know why you are so angree, since I didnt get half of your reaction from owner of the github. he was ver patient and gracieous, very rare these days! so want to thank him and you as well for responding. Update, by lowering the flash freaquency to 40MHz I was able to program this chineese camera kit. unfortunatly it is not reading the files from SD card. So I was only able to install the basic esp32 camera software on it and not S60sc. but the basic software is working very well on this module and it is faster than ESP32 becuase it has more ram, the immage quality is much higher than ov2640. Only issue with ov5640 I heared was overheating, I set the xclock frequency on the camera to 12MHz and it works very well. No over heat, no distorded image afte at least 3H of testing. they must have done something right. only problem is cooling the ESP chip that they left bare with no heatsink. that littel suck is hotest thing on the board. I think this is a good board, I wish there was more documentation on it. So I can configure the SD card on it.

closed 2025-07-02 by sinole1987