s60sc/ESP32-CAM_MJPEG2SD #625: M5Stack UnitCAMS3

splitn2, 2025-07-15:
Hi there. I am compiling with the latest IDE and libraries and managed to successfully compile the code, however I am getting panic/crash and reboot. The fault indicates some camera module connectivity.

I will put logs and configs onto this issue report later when I am near the system again.

One challenge I had in getting this running was that the SD card socket pins were not defined, I added CLK MISO and MOSI and I can see when it boots the serial output shows the SD has mounted and initial files are created, so I passed that hurdle.

In the camera_pins.h file I have verified the correct pins are used, when comparing against the M5STACK documentation, and I have selected the correct model in the appGlobals.h file......

#elif defined(CAMERA_MODEL_M5STACK_CAMS3_UNIT)
#define CAM_BOARD "CAMERA_MODEL_M5STACK_CAMS3_UNIT"
#define PWDN_GPIO_NUM  -1
#define RESET_GPIO_NUM 21
#define XCLK_GPIO_NUM  11
#define SIOD_GPIO_NUM  17
#define SIOC_GPIO_NUM  41

#define Y9_GPIO_NUM    13
#define Y8_GPIO_NUM    4
#define Y7_GPIO_NUM    10
#define Y6_GPIO_NUM    5
#define Y5_GPIO_NUM    7
#define Y4_GPIO_NUM    16
#define Y3_GPIO_NUM    15
#define Y2_GPIO_NUM    6
#define VSYNC_GPIO_NUM 42
#define HREF_GPIO_NUM  18
#define PCLK_GPIO_NUM  12

#define LED_GPIO_NUM 14

My camera unit works as it streams video with other ESP32 Camera firmware , so the camera unit is functional.

I am curious is this software aware the M5STACK_CAMS3 UNIT 5MB camera is a quite unique PY260 sensor module, not the OV2640 / OV5640 - I can't find much information on this PY260 sensor?  I am starting to wonder if this is my problem..........

So no expectation for anyone to respond yet I just putting down my thoughts and will put logs and configs in here , at that point it would be great to have some help please.

Thanks, Richard Harcourt

s60sc, 2025-07-17:
The PY260 sensor is not in the list supported by [espressif camera library](https://github.com/espressif/esp32-camera)

> My camera unit works with other ESP32 Camera firmware 

The module [website](https://docs.m5stack.com/en/unit/Unit-CAMS3%205MP) provides its own firmware and states: 

> The Unit CamS3-5MP camera is not compatible with programs for the Unit CamS3.

splitn2, 2025-07-20:
Further info, the guy who submitted the commit for MEGA_CCM works for ARDUCAM

Then I picked off the sticky tape glued on sensor and took some photos, see its actually got ARDUCAM screen printed on to it.

[image]
[image]

And on ARDUCAM website they have info on the sensor too......
https://blog.arducam.com/mega-dvp-camera-solution/

s60sc, 2025-07-20:
Try adding the following to `appGlobals.h`:

`#define CONFIG_MEGA_CCM_SUPPORT`

splitn2, 2025-07-26:
update..............

I've reverted my M5UNITCAMS3-5MP to the original PY260 ARDUCAM module. 

I know the camera works as I tried the factory test code from M5STACK and I can stream video successfully from this configuration the camera sensor is working fine.

But when compile and load the ESP32-CAM_MPEG2SD, even with the #define CONFIG_MEGA_CCM_SUPPORT added to appGlobals.h , I get a success compile and upload **but the camera fails to initialise**. I'm as confident as I can be that the hardware is fine since it worked with the M5Stack firmware......

This is the error on boot........

18:43:42.155 -> ] File: /data/common.js, size: 48985 bytes
18:43:42.155 -> [00:03:58.759 listFolder] SD_MMC: 6.2MB used of 59.4GB
18:43:42.155 -> [00:03:58.759 loadConfig] Load config
18:43:42.155 -> [00:03:58.986 updateAppStatus] Enabling motion detection
18:43:42.155 -> [00:03:58.990 prepCam] Max frame size for 8.0MB PSRAM is QSXGA
18:43:44.415 -> [00:04:01.651 setWifiSTA] Wifi Station IP from DHCP
18:43:44.450 -> [00:04:01.668 WARN startWifi] SSID  not connected not available, use AP 
18:43:44.450 -> [00:04:01.669 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_94DB1BEABF8C started, use 'http://192.168.4.1' to connect
18:43:44.450 -> [00:04:01.671 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_9.local
18:43:47.243 -> [00:04:04.483 startPing] Started ping monitoring - Off
18:43:47.243 -> [00:04:04.485 startWebServer] Starting web server on port: 80
18:43:47.243 -> [00:04:04.486 startWebServer] Remote server certificates not checked
**18:43:47.243 -> [00:04:04.486 WARN setup] Startup Failure: Camera init error 0x20002:ERROR 0x20002(131074) on CAMERA_MODEL_M5STACK_CAMS3_UNIT** 

So I'm a bit stumped with this now unsure how to fix it. Appreciate any further suggestions you might have ?

Thanks
RIchard

Also as a tidy up I added MMC CS chip select to my config so this is my full definition in camera_pins.h now.............. not that it helps with the camera issue but I guess should be there........

#elif defined(CAMERA_MODEL_M5STACK_CAMS3_UNIT)
#define CAM_BOARD "CAMERA_MODEL_M5STACK_CAMS3_UNIT"
#define PWDN_GPIO_NUM  -1
#define RESET_GPIO_NUM 21
#define XCLK_GPIO_NUM  11
#define SIOD_GPIO_NUM  17
#define SIOC_GPIO_NUM  41

#define Y9_GPIO_NUM    13
#define Y8_GPIO_NUM    4
#define Y7_GPIO_NUM    10
#define Y6_GPIO_NUM    5
#define Y5_GPIO_NUM    7
#define Y4_GPIO_NUM    16
#define Y3_GPIO_NUM    15
#define Y2_GPIO_NUM    6
#define VSYNC_GPIO_NUM 42
#define HREF_GPIO_NUM  18
#define PCLK_GPIO_NUM  12

#define LED_GPIO_NUM 14

#define SD_MMC_CLK 39 // SCLK
#define SD_MMC_CMD 38 // MOSI
#define SD_MMC_D0 40  // MISO
#define SD_MMC_CS 9 //Chip select

s60sc, 2025-07-26:
In `appGlobals.h`, set `#define DBG_ON true` and compile with Arduino Core Debug Level: Verbose which may output additional clues.

Also try PY260 with the ESP CameraWebServer example sketch.

splitn2, 2025-07-26:
A lot more verbose, and a bit more camera module specific output that I have **BOLDED**......

I will try the other sketch also and report back.

Thanks again, Richard

3:09:37.997 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein=========== Before Setup Start ===========
23:09:37.997 -> Chip Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Model             : ESP32-S3
23:09:37.997 ->   Package           : 0
23:09:37.997 ->   Revision          : 0.02
23:09:37.997 ->   Cores             : 2
23:09:37.997 ->   CPU Frequency     : 240 MHz
23:09:37.997 ->   XTAL Frequency    : 4 Features Bitfield : 0x00000012
23:09:37.997 ->   Embedded Flash    : No
23:09:37.997 ->   Embedded PSRAM    : No
23:09:37.997 ->   2.4GHz WiFi       : Yes
23:09:37.997 ->   Classic BT        : No
23:09:37.997 ->   BT Low Energy     : Yes
23:09:37.997 ->   IEEE 802.15.4     : No
23:09:37.997 -> ------------------------------------------
23:09:37.997 -> INTERNAL Memory Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Total Size        :   303912 B ( 296.8 KB)
23:09:37.997 ->   Free Bytes        :   266060 B ( 259.8 KB)
23:09:37.997 ->   Allocated Bytes   :    33540 B (  32.8 KB)
23:09:37.997 ->   Minimum Free Bytes:   262020 B ( 255.9 KB)
23:09:37.997 ->   Largest Free Block:   196596 B ( 192.0 KB)
23:09:37.997 -> ------------------------------------------
23:09:37.997 -> SPIRAM Memory Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Total Size        :  8388608 B (8192.0 KB)
23:09:37.997 ->   Free Bytes        :  8386088 B (8189.5 KB)
23:09:37.997 ->   Allocated Bytes   :      144 B (   0.1 KB)
23:09:38.181 ->   Minimum Free Bytes:  8386088 B (8189.5 KB)
23:09:38.181 ->   Largest Free Block:  8257524 B (8064.0 KB)
23:09:38.181 ->   Bus Mode          : OPI3:09:37.997 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein=========== Before Setup Start ===========
23:09:37.997 -> Chip Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Model             : ESP32-S3
23:09:37.997 ->   Package           : 0
23:09:37.997 ->   Revision          : 0.02
23:09:37.997 ->   Cores             : 2
23:09:37.997 ->   CPU Frequency     : 240 MHz
23:09:37.997 ->   XTAL Frequency    : 4 Features Bitfield : 0x00000012
23:09:37.997 ->   Embedded Flash    : No
23:09:37.997 ->   Embedded PSRAM    : No
23:09:37.997 ->   2.4GHz WiFi       : Yes
23:09:37.997 ->   Classic BT        : No
23:09:37.997 ->   BT Low Energy     : Yes
23:09:37.997 ->   IEEE 802.15.4     : No
23:09:37.997 -> ------------------------------------------
23:09:37.997 -> INTERNAL Memory Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Total Size        :   303912 B ( 296.8 KB)
23:09:37.997 ->   Free Bytes        :   266060 B ( 259.8 KB)
23:09:37.997 ->   Allocated Bytes   :    33540 B (  32.8 KB)
23:09:37.997 ->   Minimum Free Bytes:   262020 B ( 255.9 KB)
23:09:37.997 ->   Largest Free Block:   196596 B ( 192.0 KB)
23:09:37.997 -> ------------------------------------------
23:09:37.997 -> SPIRAM Memory Info:
23:09:37.997 -> ------------------------------------------
23:09:37.997 ->   Total Size        :  8388608 B (8192.0 KB)
23:09:37.997 ->   Free Bytes        :  8386088 B (8189.5 KB)
23:09:37.997 ->   Allocated Bytes   :      144 B (   0.1 KB)
23:09:38.181 ->   Minimum Free Bytes:  8386088 B (8189.5 KB)
23:09:38.181 ->   Largest Free Block:  8257524 B (8064.0 KB)
23:09:38.181 ->   Bus Mode          : OPI
23:09:38.181 -> ------------------------------------------
[... 111 lines cut ...]
23:09:44.211 ->   Largest Free Block:   122868 B ( 120.0 KB)
23:09:44.211 -> ------------------------------------------
23:09:44.211 -> SPIRAM Memory Info:
23:09:44.211 -> ------------------------------------------
23:09:44.211 ->   Total Size        :  8388608 B (8192.0 KB)
23:09:44.211 ->   Free Bytes        :  8209304 B (8016.9 KB)
23:09:44.211 ->   Allocated Bytes   :   170256 B ( 166.3 KB)
23:09:44.211 ->   Minimum Free Bytes:  4292472 B (4191.9 KB)
23:09:44.211 ->   Largest Free Block:  8126452 B (7936.0 KB)
23:09:44.211 -> ------------------------------------------
23:09:44.211 -> GPIO Info:
23:09:44.211 -> ------------------------------------------
23:09:44.211 ->   GPIO : BUS_TYPE[bus/unit][chan]
23:09:44.211 ->   --------------------------------------  
23:09:44.211 ->     19 : USB_DM
23:09:44.211 ->     20 : USB_DP
23:09:44.211 ->     38 : SDMMC_CMD
23:09:44.211 ->     39 : SDMMC_CLK
23:09:44.211 ->     40 : SDMMC_D0
23:09:44.211 ->     43 : UART_TX[0]
23:09:44.211 ->     44 : UART_RX[0]
23:09:44.211 -> ============ After Setup End =============
23:09:44.245 -> [03:52:06.443 loop] =============== Total tasks: 14 ===============
23:09:44.245 -> 
23:09:49.224 -> [03:52:11.416 resetWatchDog] WatchDog started using task: ping

s60sc, 2025-07-27:
Looking at the M5 code it appears the PY260 does not support larger frame sizes which could be the cause of:
`23:09:39.558 -> E (2350) camera: Failed to set frame size`

In `mjpeg2sd.cpp` function `prepCam()`, add:

```
#ifdef CAMERA_MODEL_M5STACK_CAMS3_UNIT
  maxFS = FRAMESIZE_QVGA;
#endif
```
to the end of:

```
  bool res = false;
  // buffer sizing depends on psram size (2M, 4M or 8M)
  // FRAMESIZE_QSXGA = 1MB, FRAMESIZE_UXGA = 375KB (as JPEG)
  maxFS = FRAMESIZE_SVGA; // 2M
  if (ESP.getPsramSize() > 5 * ONEMEG) maxFS = FRAMESIZE_QSXGA; // 8M
  else if (ESP.getPsramSize() > 3 * ONEMEG) maxFS = FRAMESIZE_UXGA; // 4M
```

s60sc, 2025-07-29:
try v10.7

closed 2025-07-29 by s60sc