s60sc/ESP32-CAM_MJPEG2SD #628: Problems with the v3.3.0 of espressif/arduino-esp32... 'jpg_scale_t' has not been declared`

EagleExe, 2025-07-28:
The newest version of **espressif/arduino-esp32** dropped about 5 days ago. Here's what's not working. I downgraded **espressif/arduino-esp32** back to v3.2.1 for now.

`C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp:61:71: error: 'jpg_scale_t' has not been declared
   61 | static bool jpg2rgb(const uint8_t* src, size_t src_len, uint8_t* out, jpg_scale_t scale);
      |                                                                       ^~~~~~~~~~~
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp: In function 'bool checkMotion(camera_fb_t*, bool, bool)':
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp:186:54: error: 'jpg_scale_t' was not declared in this scope
  186 |   if (!jpg2rgb((uint8_t*)fb->buf, fb->len, rgb_buf, (jpg_scale_t)scaling)) return motionStatus;
      |                                                      ^~~~~~~~~~~
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp: At global scope:
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp:369:71: error: 'jpg_scale_t' has not been declared
  369 | static bool jpg2rgb(const uint8_t* src, size_t src_len, uint8_t* out, jpg_scale_t scale) {
      |                                                                       ^~~~~~~~~~~
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp: In function 'bool jpg2rgb(const uint8_t*, size_t, uint8_t*, int)':
C:\Users\#####\Desktop\NEWEST\A\ESP32-CAM_MJPEG2SD\motionDetect.cpp:376:19: error: 'esp_jpg_decode' was not declared in this scope; did you mean 'esp_jpeg_decode'?
  376 |   esp_err_t res = esp_jpg_decode(src_len, scale, _jpg_read, _rgb_write, (void*)&jpeg);
      |                   ^~~~~~~~~~~~~~
      |                   esp_jpeg_decode
exit status 1

Compilation error: 'jpg_scale_t' has not been declared`

s60sc, 2025-07-28:
Thanks, espressif have changed the jpeg decoder so will need investigation

s60sc, 2025-07-29:
fixed in v10.7