s60sc/ESP32-CAM_MJPEG2SD #631: [00:00:32.160 WARN sendChunks] Failed to send /data/MJPEG2SD.htm to browser

dame333, 2025-08-02:
Hello, 
Can someone please help me with this issue? I'm getting the [00:00:32.160 WARN sendChunks] Failed to send /data/MJPEG2SD.htm to browser 
warning as well as random other warnings. The app won't load up and when i do occasionally get the app up the camera doesn't work. I have tried a couple different power sources and formatted and refreshed the data file on the sd a couple times now. Help would be appreciated, I am new to coding so please keep it simple, thanks!

s60sc, 2025-08-03:
need to provide full start up log

closed 2025-08-09 by s60sc

wangsiw1, 2025-08-26:
Hi, got this error too, also on a ESP32 S3 CAM FREENOVE board before. Log as below:

[00:00:00.509 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.510 logSetup] =============== ESP-CAM_MJPEG 10.7.3 ===============
[00:00:00.520 boardInfo] Chip ESP32-D0WD-V3, 2 cores @ 240Mhz, rev 3
[00:00:00.520 boardInfo] Flash 4.0MB, mode QIO @ 80Mhz
[00:00:00.531 boardInfo] PSRAM 4.0MB, mode QSPI @ 80Mhz
[00:00:00.531 logSetup] Compiled with arduino-esp32 v3.3.0
[00:00:00.541 printResetReason] Power on reset
[00:00:00.541 printWakeupReason] Wakeup by reset
[00:00:00.542 setup] Selected board CAMERA_MODEL_AI_THINKER
[00:00:00.762 infoSD] SD card type SDHC, Size: 29.7GB, using 1 bit mode @ 40MHz
[00:00:00.762 listFolder] Sketch size 1.6MB
[00:00:01.072 listFolder] File: /data/configs.txt, size: 7839 bytes
[00:00:01.074 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[00:00:01.076 listFolder] File: /data/common.js, size: 49333 bytes
[00:00:01.087 listFolder] File: /data/MJPEG2SD.htm, size: 100KB
[00:00:01.088 listFolder] SD_MMC: 20.7MB used of 29.7GB
[00:00:01.098 loadConfig] Load config
[00:00:01.377 updateAppStatus] Enabling motion detection
[00:00:01.382 prepCam] Max frame size for 4.0MB PSRAM is UXGA
[00:00:01.652 prepCam] Camera model OV2640 ready @ 20MHz
[00:00:01.861 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.886 WARN startWifi] SSID not connected not available, use AP
[00:00:01.888 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_186ADC2F2B14 started, use 'http://192.168.4.1' to connect
[00:00:05.314 startPing] Started ping monitoring - On
[00:00:05.318 startWebServer] Starting web server on port: 80
[00:00:05.318 startWebServer] Remote server certificates not checked
[00:00:05.329 startSustainTasks] Started 1 sustain tasks
[00:00:05.351 updateAppStatus] Enabling motion detection
[00:00:05.458 prepRecording] To record new AVI, do one of:
[00:00:05.458 prepRecording] - press Start Recording on web page
[00:00:05.458 prepRecording] - move in front of camera

[00:00:05.469 checkMemory] Setup Free: heap 86844, block: 65524, min: 86844, pSRAM 2102764
[00:00:05.479 loop] =============== Total tasks: 17 ===============

[00:00:07.739 isNight] Night time
[00:00:10.316 resetWatchDog] WatchDog started using task: ping
[00:00:25.931 onWiFiEvent] WiFi AP client connection
[00:00:33.312 WARN sendChunks] Failed to send to browser: /data/MJPEG2SD.htm, err ESP_ERR_HTTPD_RESP_SEND
[00:01:10.376 logSetup] Setup RAM based log, size 7168, starting from 2342

[00:01:10.377 logSetup] =============== ESP-CAM_MJPEG 10.7.3 ===============
[00:01:10.387 boardInfo] Chip ESP32-D0WD-V3, 2 cores @ 240Mhz, rev 3
[00:01:10.387 boardInfo] Flash 4.0MB, mode QIO @ 80Mhz
[00:01:10.398 boardInfo] PSRAM 4.0MB, mode QSPI @ 80Mhz
[00:01:10.398 logSetup] Compiled with arduino-esp32 v3.3.0
[00:01:10.408 printResetReason] Reset due to task watchdog
[00:01:10.408 printWakeupReason] Wakeup by reset
[00:01:10.419 setup] Selected board CAMERA_MODEL_AI_THINKER
[00:01:10.478 infoSD] SD card type SDHC, Size: 29.7GB, using 1 bit mode @ 40MHz
[00:01:10.478 listFolder] Sketch size 1.6MB
[00:01:10.789 listFolder] File: /data/configs.txt, size: 7839 bytes
[00:01:10.790 listFolder] File: /data/Auxil.htm, size: 46254 bytes
[00:01:10.792 listFolder] File: /data/common.js, size: 49333 bytes
[00:01:10.804 listFolder] File: /data/MJPEG2SD.htm, size: 100KB
[00:01:10.804 listFolder] SD_MMC: 20.7MB used of 29.7GB
[00:01:10.815 loadConfig] Load config
[00:01:11.093 updateAppStatus] Enabling motion detection
[00:01:11.098 prepCam] Max frame size for 4.0MB PSRAM is UXGA
[00:01:11.368 prepCam] Camera model OV2640 ready @ 20MHz
[00:01:11.578 setWifiSTA] Wifi Station IP from DHCP
[00:01:11.629 WARN startWifi] SSID not connected not available, use AP
[00:01:11.632 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_186ADC2F2B14 started, use 'http://192.168.4.1' to connect
[00:01:15.059 startPing] Started ping monitoring - On
[00:01:15.062 startWebServer] Starting web server on port: 80
[00:01:15.063 startWebServer] Remote server certificates not checked
[00:01:15.073 WARN setup] Startup Failure: Crash loop detected, check log
[00:01:15.073 loop] =============== Total tasks: 14 ===============

[00:01:15.097 onWiFiEvent] WiFi AP client connection
[00:01:20.060 resetWatchDog] WatchDog started using task: ping
[00:08:14.723 onWiFiEvent] WiFi AP client connection

dame333, 2025-08-26:
I resolved the issue by changing the power source,  re inserting the sd
card and resetting the esp cam. I have multiple cameras and this issue
happens frequently. I have also formatted and re downloaded the data file
to the sd card. I think the problem stems from the sd card used. Get a high
quality sd and it will help. Good luck!

dame333, 2025-08-26:
> Thanks for replying! It seems to be something wrong with the wifi on my
> phone as suggested in #647
> <https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/647>. I also did the
> sd card re formatting as you and the other issue mentioned, which may also
> helped. Now it seems to be working fine.

I forgot to mention that it works best on google chrome. Glad to hear you
got it working. It can be frustrating.