s60sc/ESP32-CAM_MJPEG2SD #632: No Audio

maiRAGE, 2025-08-04:
I'm running a xiao esp32s3 sense board and I just can't get the audio to stream from the esp to the web browser and out of my speakers and I'm not sure if it's something I'm doing wrong, my setup, or if maybe a library or something else has been updated and is the cause. I've tried different switches editing the code erasing parts of it and also tried over https(couldn't get https to work either but that'll be my next focus after audio) etc..  I was really trying to not ask and to figure it out myself but at this point I've tried for a while with no success.  So can I get some help with this please?  (also I started from scratch with new installs of the code so other than the snippets the code should be how it is at the time of download.) The one thing I do know is that it keeps looking like it could be related to the javascript (which I also tried to adjust before reverting it) but I'm not very advanced at coding and such yet so I can't be sure.

[image]

[image]

[image]

[image]

[image]

s60sc, 2025-08-04:
Please provide full startup log

You shouldnt need the code - what browser are you using.

s60sc, 2025-08-05:
ignore `not allowed to load local resource: file:///C:.....`

`uncaught (in promise) TypeError: Cannot read properties of undefined (reading addModule) ` might be because your browser or device does not support `AudioWorklet`.  
In `async function runSpkr(index) `, try adding:
`if (!audioContextSpkr.audioWorklet) console.error('AudioWorklet not supported in this browser/environment');`
before:
`await audioContextSpkr.audioWorklet.addModule('data:text/javascript;base64,' + btoa(audioWorkletScript));`

maiRAGE, 2025-08-06:
so good new is that I got eh error that stated it wasn't supported by the browser/environment so I checked the caniuse website to check chrome versions it worked with as well as the version I'm running which can be seen in the images as well as the error.  I then seen that audioworklet can run without https which I'm not doing at the moment because it won't run for me for some reason even after I clicked the switch put in the esp ip addr as well as the router addr and putting the dns addr as my router ip. but still nothing.  I also put in my cert and key in the box's for them in the cert.cpp.  and imported them to the browsers but the esp itself is failing to start the https server before I even get a browser error other than it being unable to connect.

[image]

[image]
[image]

[image]

[image]

[image]
[image]
[image]

maiRAGE, 2025-08-06:
Is there another way I can set it up other than the ui then? Because I've noticed that even when I change the ssid and password for another in the ui it doesn't work and just using the ap is the only way. From the configuration it seems like everything may be stored in the memory though.

s60sc, 2025-08-06:
I've noticed that espressif have changed the way they handle certificates in the https server library so I'll investigate

s60sc, 2025-08-09:
https cerificate handling updated in v10.7.2 following espressif lib changes. Still has stability issues

maiRAGE, 2025-08-10:
well it still didn't work for me but I like how much smother it seems now to check it than it was.

I tried both this format in the servercert.pem/prvtkey.pem but neither worked for me yet likely because of the stability issue you're talking about I guess. but I wasn't sure which format below to try so I tried both but neither worked. (I'm assuming the second seeing as the code seems to be looking for a file now rather than a variable.)

const char* servercert = R"~(
-----BEGIN CERTIFICATE-----
MIIFIjCCAwqgAwIBAgIUbQMOqZzrebYe
******************************************
stbOAksjByQCv7UL5EvbUJkEFh0LQg==
-----END CERTIFICATE-----
)~";

-----BEGIN CERTIFICATE-----
MIIFIjCCAwqgAwIBAgIUbQMOqZzrebYe
******************************************
stbOAksjByQCv7UL5EvbUJkEFh0LQg==
-----END CERTIFICATE-----

s60sc, 2025-08-10:
As per new instructions in certificates.cpp, ota the created cert files to esp

maiRAGE, 2025-08-10:
I did.  it seemed like I had to do them seperately but it didn't work.  I didn't share the entire cert and key above but I was making sure it was still in that format.  maybe I made the files in the wrong format are they supposed to be saved as a .c file?

s60sc, 2025-08-10:
The problem is that running https with live AV streaming is asking too much of a little $5 MCU

maiRAGE, 2025-08-11:
oh gotcha that makes sense.  By the way before I found your code I was actually working on my own and was making it pretty similar to what you have done (except at a much lower level).  I actually made mine so that there was an audio trigger with mine so that it would automatically start recording audio/video once a certain threshold was made and would extend the recording if it was triggered during the recording session.  do you think you may want to try and incorporate something like that into this code? I think it would be an excellent addition to the code. it would use the ulp that the esp32 has so it can be in low power mode this way it won't have to reboot everything upon the trigger and can start recording immediatly.