s60sc/ESP32-CAM_MJPEG2SD #638: can't crete or read file in sd card

TwoBrushes, 2025-08-11:
platform =   https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-2/platform-espressif32.zip

log output:

[00:00:02.048 logSetup] =============== ESP-CAM_MJPEG 10.7.3 ===============
[00:00:02.050 boardInfo] Chip ESP32-S3, 2 cores @ 240Mhz, rev 0
[00:00:02.050 boardInfo] Flash 8.0MB, mode QIO @ 80Mhz
[00:00:02.051 boardInfo] PSRAM 8.0MB, mode OPI @ 80Mhz
[00:00:02.051 logSetup] Compiled with arduino-esp32 v3.3.0
[00:00:02.051 printResetReason] Power on reset
[00:00:02.594 infoSD] SD card type SDHC, Size: 3.8GB, using 1 bit mode @ 40MHz
[00:00:02.595 listFolder] Sketch size 1.6MB
[00:00:02.729 listFolder] SD_MMC: 247.0MB used of 3.8GB
[00:00:02.730 loadConfig] Load config
[00:00:02.733 WARN checkConfigFile] Failed to create file /data/configs.txt
[00:00:02.735 WARN checkConfigFile] Failed to create file /data/configs.txt
[00:00:02.894 setWifiSTA] Wifi Station IP from DHCP
[00:00:02.899 WARN startWifi] SSID  not connected not available, use AP
[00:00:02.901 WARN setupMdnsHost] mDNS host:  Failed
[00:00:06.715 startPing] Started ping monitoring - On
[00:00:06.717 startWebServer] Starting web server on port: 80
[00:00:06.717 startWebServer] Remote server certificates not checked
[00:00:06.717 WARN setup] Startup Failure: No file: /data/configs.txt
[00:00:06.717 loop] =============== Total tasks: 14 ===============

[00:00:11.715 resetWatchDog] WatchDog started using task: ping

rotifull, 2025-08-11:
Did you add the data folder to the sd?

TwoBrushes, 2025-08-11:
> Did you add the data folder to the sd?

Yes, I have copied the data directory to the sd card， ，This project works fine under platform-espressif32 53.03.13.

rotifull, 2025-08-11:
Ok are the sd card pins correct or had you modified them? An other problem is if you set like an servo to an pin with the sd card uses and then it won't work, it had also happened to me

TwoBrushes, 2025-08-12:
> Sorry I just thought I could help from my own experience and I am not the developer of this code so I don't know much about this

Still need to thank you.

kjkent, 2025-08-19:
Same issue here with the Freenove ESP32-S3-WROOM-1-N16R8 with OV3660, using the same PlatformIO `platform = https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-2/platform-espressif32.zip`.

Full debug log below:

<details>

```text
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1180
load:0x403c8700,len:0xc2c
load:0x403cb700,len:0x311c
entry 0x403c88b8
[     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202fdf4
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202fdc0
[    24][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202fd8c
[    35][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202fd58
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202fdf4
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202fdc0
[    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202fd8c
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202fd58
[    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202fdf4
[   103][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202fdc0
[   114][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202fd8c
[   125][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202fd58
[   149][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc9a6d4
[   160][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc9a6d4
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   289372 B ( 282.6 KB)
  Free Bytes        :   252044 B ( 246.1 KB)
  Allocated Bytes   :    33064 B (  32.3 KB)
  Minimum Free Bytes:   246648 B ( 240.9 KB)
  Largest Free Block:   188404 B ( 184.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8386096 B (8189.5 KB)
  Allocated Bytes   :      136 B (   0.1 KB)
  Minimum Free Bytes:  8386096 B (8189.5 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
[... 303 lines cut ...]
  Minimum Free Bytes:   157332 B ( 153.6 KB)
  Largest Free Block:    94196 B (  92.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  4711928 B (4601.5 KB)
  Allocated Bytes   :  3667296 B (3581.3 KB)
  Minimum Free Bytes:  4711928 B (4601.5 KB)
  Largest Free Block:  4587508 B (4480.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    38 : SDMMC_CMD
    39 : SDMMC_CLK
    40 : SDMMC_D0
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
[00:00:05.095 loop] =============== Total tasks: 18 ===============
```

</details>

s60sc, 2025-08-22:
I dont use platformio so I cant help